matthijskooijman/arduino-lmic #111: Unable to send packet

Roberto6969, 2017-08-10:
Is there any way to debug sending process? I have problem with two almost identical sensors and SW. In loop is:
void loop() {
  do_send(&sendjob);
  os_runloop();
}

at the end of do_send(osjob_t j) is 
    LMIC_setTxData2(1, message.getBytes(), message.getLength(), 0);    
    Serial.println(F("Packet queued"));

So the problem is the sending process stops with the message:
Ready to send message of 1533 bytes size and 8 bytes length.
Packet queued

How can i find out where the sending process hangs?

Thank you!

matthijskooijman, 2017-08-18:
Did you try the `LMIC_PRINTF_TO` and `LMIC_LOGLEVEL` macros (they might be called slightly differently) in config.h?

Also, if you're trying to send a 1533 byte message, that's your problem - LMIC has a max message size of 50 bytes or something (64 bytes minus headers) IIRC.

Roberto6969, 2017-08-28:
Actually i don't send so much data. Here is th code which shows 1363 bytes of data but it is not true - as you can see i sending only ".addUint16(p);" and this could not be so long:
// Prepare upstream data transmission at the next possible time.

    LoraMessage message; // clear 
    readVoltage();
    readMeteo(); // reading data from BMP280
    message
    //.addLatLng(-33.905052, 151.26641)
    //.addUnixtime(1467632413)
    //.addUint8(10)
    //.addUint16(v) * * * * *
    //.addTemperature(t)
    //.addHumidity(h)
    .addUint16(p);
    /* to vklopiš za test 2 */
    Serial.print(F("\nReady to send message of "));
    message.getBytes();
    Serial.print(F(" bytes size and ")); Serial.print(message.getLength()); Serial.println(F(" bytes length."));

And about debugging: whenever i uncomment lines //#define LMIC_DEBUG_LEVEL 2 and //#define LMIC_PRINTF_TO Serial i receive error at compiling. Is there something i missed?

matthijskooijman, 2017-08-29:
> And about debugging: whenever i uncomment lines //#define LMIC_DEBUG_LEVEL 2 and //#define LMIC_PRINTF_TO Serial i receive error at compiling. Is there something i missed?

You'll have to tell us the error you got, we're not psychic :-)
If you're running on a non-AVR board, the commits from the non-avr-printf branch might be helpful (IIRC I wrote them, but haven't gotten suficient testing results yet for merging that branch).

Roberto6969, 2017-08-29:
Well, you are doing great job so you could be a magician ;-)

Here is the error at the time of compiling:

In file included from C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\lmic.h:18:0,

                 from C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\oslmic.c:12:

C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\oslmic.c: In function 'os_runloop_once':

C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\oslmic.c:118:97: error: 'is_runnable' undeclared (first use in this function)

             lmic_printf("%lu: Running job %p, cb %p, deadline %lu\n", os_getTime(), j, j->func, is_runnable ? 0 : j->deadline);

                                                                                                 ^

C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\oslmic.h:250:58: note: in definition of macro 'lmic_printf'

     #define lmic_printf(fmt, ...) printf_P(PSTR(fmt), ## __VA_ARGS__)

                                                          ^

C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\oslmic.c:118:97: note: each undeclared identifier is reported only once for each function it appears in

             lmic_printf("%lu: Running job %p, cb %p, deadline %lu\n", os_getTime(), j, j->func, is_runnable ? 0 : j->deadline);

                                                                                                 ^

C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\oslmic.h:250:58: note: in definition of macro 'lmic_printf'

     #define lmic_printf(fmt, ...) printf_P(PSTR(fmt), ## __VA_ARGS__)

                                                          ^
So this error happens only when i uncomment those two lines. I am running this on Atmel ATMEGA328P (only chip with minimum components). This configuration has worked for some time - rouhgly before two months.

matthijskooijman, 2017-08-30:
I just fixed the compilation problem with `LMIC_DEBUG_LEVEL` > 1 in git master.