matthijskooijman/arduino-lmic #113: How to test with single channel "gateway"?

skandragon, 2017-08-17:
I have a need to run a small (aka cheap) single channel gateway that could very well be destroyed at any moment, hence the need for cheap.

I've locked this gateway down to a single frequency and data rate, and would like to configure LMIC to use only this channel, either with ABP, or ideally OTAA.  I've tried the example sketches for both, but the gateway does not seem to see the transmissions, if they are even happening.

I've tried to add more debugging by enabling the radio logging (define debug as 2 rather than 0) but the code failed to compile; I've not yet had time to dig into things there.

Is there an example somewhere of using a single channel forwarder with TTN, either with OTAA or ABP?

skandragon, 2017-08-17:
I should note that I'm wanting to use a US (NA) frequency.

dragino, 2017-08-29:
You can check here for the examples for how to use this library with single channel gateway. 
[http://wiki.dragino.com/index.php?title=Connect_to_TTN](url)

skandragon, 2017-09-13:
That example only shows using ABP, which does work -- but is not as attractive as OTAA.

I see my single channel gateway transmitting, but it seems the node doesn't receive.  Also, it is still cycling through frequencies:

```63691: TXMODE, freq=902300000, len=23, SF=7, BW=125, CR=4/5, IH=0
67954: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
439103: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
450920: TXMODE, freq=903000000, len=23, SF=8, BW=500, CR=4/5, IH=0
763159: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
824307: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
1079896: TXMODE, freq=913100000, len=23, SF=8, BW=125, CR=4/5, IH=0
1396280: RXMODE_SINGLE, freq=926900000, SF=8, BW=500, CR=4/5, IH=0
1458524: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
1508949: TXMODE, freq=912600000, len=23, SF=8, BW=500, CR=4/5, IH=0
```

avbentem, 2017-10-14:
https://www.thethingsnetwork.org/forum/t/can-lmic-1-6-be-set-up-to-use-a-single-channel-and-sf/5207/11 might help, but only supports OTAA when it succeeds in its very first try, and was not tested for US. (Please respond there as well if you figured out if changes are needed.)