matthijskooijman/arduino-lmic #115: Draws a lot of current

Fireobridy, 2017-08-20:
Can you please implement a sleep mode, so the system does not draw so much current when waiting for the next tx?

Oliv4945, 2017-08-21:
Hi, the radio is already in sleep mode and you can put the Arduino in sleep thanks to this library
https://github.com/rocketscream/Low-Power

Aayk, 2018-01-17:
Oliv4945, sorry I don't undertand. So does the LMIC library put the radio always in sleep mode and when it receive and transmit wake up?

Aayk, 2019-04-12:
> 
> Hi, the radio is already in sleep mode and you can put the Arduino in sleep thanks to this library https://github.com/rocketscream/Low-Power

Hi Oliv4945. Where can I find the code about LoRa sleep? I searched in all LMIC libray but I don't found where it goes to sleep

Oliv4945, 2019-04-14:
@Aayk : You can find it [radio.c](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L822), at several places.