matthijskooijman/arduino-lmic #116: Unable to join ttn network ttn-otaa.ino

gtawde, 2017-08-21:
I am using a Arduino Pro mini 3.3V with a modtronix inAir9b device. 

Pin mapping is as follows
.nss = 10,
.rxtx = 6,
.rst = 5,
.dio = {2,3,4}

I am able to send a join request and I can also see it on my dashboard, but i get a EV_JOIN_FAILED and I cannot send any payload data cause the device only seems to be sending  activation messages.

I have tried the ttn_otaa.ino file

thayuriko, 2017-08-21:
Same here. Would be nice to know why that happens.

hallard, 2017-08-22:
Unable to join mean 3 causes 

- error in Keys (deveui, appkey, ...) 95% of errors
- device not setup in back end
- no gateway in area

aderusha, 2017-09-23:
I am seeing similar behavior with the OTAA example, but annoyingly it does in fact work maybe 1% of the time - if I leave it running over night I'll get one or two packets through.  The gateway is in the room and I've confirmed that the keys have been copied over w/ the correct byte order and are registered on TTN (which is the only way one packet would actually arrive intact).

christopherdro, 2017-09-26:
Are you sure you have the correct freq and radio setup in config? https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h#L8

Also worth trying the suggestion in this comment https://github.com/matthijskooijman/arduino-lmic/issues/55#issuecomment-260064164

aderusha, 2017-09-26:
The fact that it will every now and again successfully send data to TTN suggests strongly to me that the frequency plan, APPEUI, DEVEUI, and APPKEY are correct.  I don't see any way I would be seeing packets show up without those 4 factors being correctly configured.

hackermesh, 2017-09-27:
I am also unable to join using otaa... but i have noticed that letting the node a few minutes it reachs SF9, and then it joins properly, as you can see in this image:

[image]

Please, how could I force the node to use only SF9?
Kind regards

matthydehoog, 2018-03-17:
Is there a way to debug this.
I only receive on the serial monitor this message:
Starting
Packet queued
401: EV_JOINING

Seems no other events occure.
Did check several things already. Any one an idea where to start?

matthydehoog, 2018-03-18:
Got the same issue only also SF9 does not result in a join.
[image]

The serial monitor give me this:
`37641222: Running job 0x240, cb 0x429, deadline 37641217
37641326: engineUpdate, opmode=0xc
37642995: Uplink join pending
37644944: Airtime available at 37641342 (previously determined)
37649104: Ready for uplink
37651337: Updating info for TX at 37642994, airtime will be 12864. Setting available time for band 0 to 2901540864
37658409: TXMODE, freq=868100000, len=23, SF=9, BW=125, CR=4/5, IH=0
37666886: irq: dio: 0x0 flags: 0x8
37666967: Scheduled job 0x240, cb 0x2052 ASAP
37667952: Running job 0x240, cb 0x2052, deadline 0
37671276: Scheduled job 0x240, cb 0x205e at 37976571
37976578: Running job 0x240, cb 0x205e, deadline 37976571
37976746: RXMODE_SINGLE, freq=868100000, SF=9, BW=125, CR=4/5, IH=0
37980322: irq: dio: 0x1 flags: 0x80
37980405: Scheduled job 0x240, cb 0x22b6 ASAP
37982054: Running job 0x240, cb 0x22b6, deadline 0
37985378: Scheduled job 0x240, cb 0x2081 at 38041759
38041763: Running job 0x240, cb 0x2081, deadline 38041759
38041932: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
38054564: irq: dio: 0x1 flags: 0x80
38054646: Scheduled job 0x240, cb 0x22c9 ASAP
38055694: Running job 0x240, cb 0x22c9, deadline 0
38059010: Scheduling next join at 51523261
38061805: Scheduled job 0x240, cb 0x429 at 38061800
38065185: Running job 0x240, cb 0x429, deadline 38061800
38068885: engineUpdate, opmode=0xc
38071161: Uplink join pending
38073110: Airtime available at 51523261 (previously determined)
38077270: Uplink delayed until 51523261
38079875: Scheduled job 0x240, cb 0x429 at 51523136`

matthydehoog, 2018-03-18:
Solved.
I did use the APPKEY also with least-significant-byte first like the APPEUI and DEVEUI. But the APPKEY should be most-significant-byte first.

deepakjayan93, 2019-02-27:
I tried to connect my SX1276 (RFM95 compatible) LoRa module to THE THINGS NETWORK using OTAA activation method. I downloaded MCCI LORAWAN LMIC Library (version  2.3.2). The LoRa module was connected to arduino UNO. The APPEUI AND DEVEUI  were written into the code in little endian mode and the APPKEY in big endian mode. European frequency was set in the " lmic_project_config.h " file.  Since I am using adruino  UNO , I disabled ping and beacons . Otherwise it could not be stored in adruino UNO flash memory as mentioned in "Mobilefish  Lorawan tutorials" . No other changes were made in the code.
[image]
 I have attached the code along with the output in the serial monitor. I could not connect to TTN. The message 'EV_TXSTART' appears in the serial monitor. We waited for 40 minutes. Only this message appears in the serial monitor.
[image]
[image]

 The output appearing in the serial monitor has been attached herewith.  What could be the possible reason for not getting connected to TTN? Please reply.

pedroamfarias, 2019-03-27:
> I tried to connect my SX1276 (RFM95 compatible) LoRa module to THE THINGS NETWORK using OTAA activation method. I downloaded MCCI LORAWAN LMIC Library (version 2.3.2). The LoRa module was connected [...]

Hi! I tested what suggested by tomtor in https://www.thethingsnetwork.org/forum/t/lmic-on-adafruit-lora-feather-successfully-sends-message-to-ttn-and-then-halts-with-packet-queued/3762/30?u=pedroamfarias

He said

> The LMIC code is quite portable.
> 
> The only issue I had was when using a less accurate clock.
> 
> You could try adding
> 
> LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);

Just put "LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);" after LMIC_reset() and it's worked fine to me! my device is a LoRa32u4ii

Alars-2406, 2019-12-13:
I am trying to connect my HPD13A node to the node configured with a DHT sensor and switch. But I'm unable to upload the code to my arduino UNO. I have been receiving the error of EU-bandwith not configured with lmic, and also multiple libraries detected. Any solutions available?
[image]