matthijskooijman/arduino-lmic #118: Feather M0: constant EV_TXCOMPLETE and Packet queued?

PhatHub, 2017-09-16:
I've had the same problem as a few others with the Adafruit M0 Feather, where the serial debug showed a series of `EV_TXCOMPLETE` and `Packet queued`. For me, I'll get three of them one right after the other, and then three sets of reports for the`TXMODE` and `RXMODE_SINGLE`s, and it'll repeat over and over. 
Here's an example:

> Packet queued
> 25147489: EV_TXCOMPLETE (includes waiting for RX windows)
> Queueing next packet
> Packet queued
> 25903121: EV_TXCOMPLETE (includes waiting for RX windows)
> Queueing next packet
> Packet queued
> 26658815: EV_TXCOMPLETE (includes waiting for RX windows)
> Queueing next packet
> Packet queued
> 27414512: EV_TXCOMPLETE (includes waiting for RX windows)
> Queueing next packet
> 25083383: RXMODE_SINGLE, freq=923900000, SF=7, BW=500, CR=4/5, IH=0
> 25146243: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
> 25147506: engineUpdate, opmode=0x900
> 25772507: engineUpdate, opmode=0x908
> 25772688: TXMODE, freq=902700000, len=26, SF=7, BW=125, CR=4/5, IH=0
> 25839016: RXMODE_SINGLE, freq=924500000, SF=7, BW=500, CR=4/5, IH=0
> 25901812: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
> 25903138: engineUpdate, opmode=0x900
> 26528137: engineUpdate, opmode=0x908
> 26528319: TXMODE, freq=902900000, len=26, SF=7, BW=125, CR=4/5, IH=0
> 26594646: RXMODE_SINGLE, freq=925100000, SF=7, BW=500, CR=4/5, IH=0
> 26657507: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
> 26658834: engineUpdate, opmode=0x900
> 27283835: engineUpdate, opmode=0x908
> 27284016: TXMODE, freq=903100000, len=26, SF=7, BW=125, CR=4/5, IH=0
> 27350344: RXMODE_SINGLE, freq=925700000, SF=7, BW=500, CR=4/5, IH=0
> 27413204: RXMODE_SINGLE, freq=923300000, SF=9, BW=125, CR=4/5, IH=0
> 27414533: engineUpdatePacket queued
> 28170209: EV_TXCOMPLETE (includes waiting for RX windows)

While the above repeats over and over again, and it actually only transmits once every several hundred queued packets. (I have a sniffer on the gateway and the frame counts jump by the hundreds).

I haven't dug further into it, but I'm suspecting that maybe the `os_getTime()` function might not be grabbing or calculating the `micros()` time correctly? (The above example was supposed to transmit only every ten seconds) I'm also puzzled as to how the transmit-complete event happens so often. (Hardware error?)

However, I haven't encountered this with my Arduino Nano or the Feather 32u4 builds... they're both running smoothly. :)

This is using the "mostly stock" `ttn-abp.ino`, level 2 debugging, and the non-avr patch for the printf debugging.

Thanks!

PhatHub, 2017-09-16:
ARGH. I left it alone for a while then re-uploaded it, and it's transmitting consistently again. *sigh.* 
This time I set debugging to level zero and killed the printf defines in `config.h`.

closed 2017-09-20 by PhatHub

jerrylogansquare, 2017-11-04:
I'm seeing a similar issue.  Did setting debug to level 0 help?