matthijskooijman/arduino-lmic #119: Disable Waiting For Downlink

athtest800, 2017-09-21:
Hello!
How can I disable the waiting for downlink?
I want to send data and sleep immediately without waiting message from gateway.
Thank you!

matthijskooijman, 2017-09-21:
There is not currently any support for this. Also note that doing so would, AFAICS, break LoRaWAN compatibility: Your device will no longer be able to receive MAC commands, which might be needed for proper management of the network. I think these commands aren't used much by TTN right now (not sure about other networks), but I can imagine this will change in the future and devices that do not support them might be rejected by some networks.

avbentem, 2017-10-14:
TTN surely uses MAC commands for ADR, so disabling is indeed a no-go, I'd say.