matthijskooijman/arduino-lmic #120: How to lock the frequency when haveing a single channel gateway?

RickardPettersson, 2017-09-26:
I have a single channel gateway running at 868.1 and when i running your library i getting after set debuging on the radio that its joining at 868.1 but after that sending hello world message in other frequencys...

RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
RXMODE_SINGLE, freq=868300000, SF=10, BW=125, CR=4/5, IH=0

Any idea how to set it to only run 868100000 frquency?

jrnker, 2017-09-26:
Hmm, good question, I've been wondering about that as well. I've been trying to use `LMIC_disableChannel(x)` but that doesn't seem to have any effect.

One nasty solution is to modify your lorabase_xxnnn.h (in my case lorabase_eu868.h) to only mention the frequency you want to lock at at, like:
```
enum {
        EU868_F1 = 868100000,      // g1   SF7-12
        EU868_F2 = 868100000,      // g1   SF7-12 FSK SF7/250
        EU868_F3 = 868100000,      // g1   SF7-12
        EU868_F4 = 868100000,      // g2   SF7-12
        EU868_F5 = 868100000,      // g2   SF7-12
        EU868_F6 = 868100000,      // g3   SF7-12
        EU868_J4 = 868100000,      // g2   SF7-12  used during join
        EU868_J5 = 868100000,      // g2   SF7-12   ditto
        EU868_J6 = 868100000,      // g2   SF7-12   ditto
};
enum {
        EU868_FREQ_MIN = 868100000,
        EU868_FREQ_MAX = 868100000
}; 
``` 
..but that isn't the right way to do it. I just don't know a better way... 
Anyone have a better idea?

PhatHub, 2017-09-28:
Sorry to bring another "hackish" way to get at this... buuuut there are two fields under LMIC that you can kind of "force" a channel to be used (This is on us915 btw):
`LMIC.txChnl` and `LMIC.chRnd`. 
For example, if I want to force it to use channel 0, I write:
```
LMIC.txChnl = 0;
LMIC.chRnd = 7;
LMIC_setTxData2( blah blah blah );
```
This is after a lot of trial and error, so I'm not exactly sure why having chRnd at 7 would cause it to wrap around to 0 when it finally transmits. (btw, for some reason my "random channel hopping" is actually sequential, so YMMV)

T94T, 2018-04-25:
Try using this:

```
#define CHANNEL  1

for (uint8_t i = 0; i < 9; i++) {
  if (i != CHANNEL) {
    LMIC_disableChannel(i);
  }
}
```

Algram, 2018-05-18:
I can confirm that the method from @T94T works perfectly!

peerv, 2018-09-17:
Solution from T94T worked for me too.
If you want to transmit on eu freq 868100000 set #define CHANNEL 0

ismaelit, 2020-04-28:
> Try using this:
> 
> ```
> #define CHANNEL  1
> 
> for (uint8_t i = 0; i < 9; i++) {
>   if (i != CHANNEL) {
>     LMIC_disableChannel(i);
>   }
> }
> ```

This method works, but pay attention for the AU915-928 frequency plan for the TTN: you must use the channel index of sub-bands: 

"Note that The Things Network uses 2nd Sub-Band only (channels 8 to 15 and 65). You’ll need to program the specific channels into the devices in order to make them work with TTN." 

If you want to use channel 916.8 only, you need to disable all channels but the 8th (not the ch index 1).

jadcamille, 2021-03-15:
> Try using this:
> 
> ```
> #define CHANNEL  1
> 
> for (uint8_t i = 0; i < 9; i++) {
>   if (i != CHANNEL) {
>     LMIC_disableChannel(i);
>   }
> }
> ```

Hey where do i put this line of code if i am using this 

// MIT License
// https://github.com/gonzalocasas/arduino-uno-dragino-lorawan/blob/master/LICENSE
// Based on examples from https://github.com/matthijskooijman/arduino-lmic
// Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman

#include <lmic.h>
#include <hal/hal.h>

/*************************************
 * TODO: Change the following keys
 * NwkSKey: network session key, AppSKey: application session key, and DevAddr: end-device address
 *************************************/
static const u1_t NWKSKEY[16] = {gdgdgdgdg, 0x32, 0x08 };
static const u1_t APPSKEY[16] = { 0dfgdfgdfgdfg0, 0x79, 0x35, 0dfgdfgdfgdfg0xCE, 0x8F };
static const u4_t DEVADDR = 0x2dfgdfg78;

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui (u1_t* buf) { }
void os_getDevEui (u1_t* buf) { }
void os_getDevKey (u1_t* buf) { }

static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 6, 7},
};

void onEvent (ev_t ev) {
    if (ev == EV_TXCOMPLETE) {
        Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
        // Schedule next transmission
        os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
    }
}

void do_send(osjob_t* j){
    // Payload to send (uplink)
    static uint8_t message[] = "Truck Track";

    // Check if there is not a current TX/RX job running
    if (LMIC.opmode & OP_TXRXPEND == 0) {
        Serial.println(F("OP_TXRXPEND, not sending"));
    } else {
        // Prepare upstream data transmission at the next possible time.
        LMIC_setTxData2(1, message, sizeof(message)-1, 0);
        Serial.println(F("Sending uplink packet..."));
    }
    // Next TX is scheduled after TX_COMPLETE event.
}

void setup() {
    Serial.begin(115200);
    Serial.println(F("Starting..."));

    // LMIC init
    os_init();

    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    // Set static session parameters.
    LMIC_setSession (0x1, DEVADDR, NWKSKEY, APPSKEY);

    // Disable link check validation
    LMIC_setLinkCheckMode(0);

    // TTN uses SF9 for its RX2 window.
    LMIC.dn2Dr = DR_SF9;

    // Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
    LMIC_setDrTxpow(DR_SF7,14);

    // Start job
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}