matthijskooijman/arduino-lmic #121: Removing the limitation on number of transmissions imposed by lmic

sajanikapukotuwa, 2017-10-01:
To the best of my knowledge, this lmic library limits the number of transmissions that can be done by a single node. I have set up a private LoRa server and would like to remove this limitation from all nodes to perform a scientific experiment. Would really appreciate if someone can shred some light as to how it can be done. Thank you.

matthijskooijman, 2017-10-02:
These limitations are based on government regulations, so removing them would make your node an illegal transmitter AFAIU.

If you insist, you can reset the duty cycle limits (and allow a message to be sent immediately) by doing:

```
    LMIC.bands[BAND_MILLI].avail =
    LMIC.bands[BAND_CENTI].avail =
    LMIC.bands[BAND_DECI ].avail = os_getTime();
```
You'd need to do this before or after every packet sent.

closed 2017-10-02 by matthijskooijman

sajanikapukotuwa, 2017-10-02:
Thank you very much. I have taken down the changes required. Not to worry, this is only to conduct an experiment. When we deploy, we will never use this many transmissions. Our nodes do not have that much energy anyway. Thanks again.

matthijskooijman, 2017-10-02:
I'm not aware of any exceptions for experiments, so technically your experiment will be illegal AFAIU. It's up to you to decide if that's acceptable.

sajanikapukotuwa, 2017-10-02:
I think there is a very special shielded room in a lab within the university. I can conduct the experiment within that.  Hope that is fine?

matthijskooijman, 2017-10-02:
Yeah, that should be perfect, AFAIK :-)