matthijskooijman/arduino-lmic #122: Unable to receive (and JOINING properly)

tyboth, 2017-10-09:
I have 2 lora modules. The lora shield Dragino on an arduino uno and a Leather M0 lora (I used the 2 branches). I can't receive ttn gateway but I got a free trial period from an operator (Objenious). 

When I try to JOIN the network the request is well received by the server. Then a "JOIN ACCEPT" is sent to my device but neither the shield or the leather received this event. 

When I run otaa code I only have:
```
Starting
Packet queued
233: EV_JOINING
```

I think that there is a link with the fact the network I use works only with ADR.

I tried this:     
```
// LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();
    
    LMIC_setLinkCheckMode(1);
    LMIC_setAdrMode(1);
    LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100);

    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
```

When I tried to communicate with APB (it's not supposed to works in my network) just after trying to JOIN with OTAA the network received uplinks from the device and after each one, emitted a downlink not received by the devices. The 2 devices have different antennas and anyway the operator website say that I have more than 50 antennas in a radius of 20km (I'm in Paris). A raw communication between modules works.

My pin mapping for the shield:
```
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 6, 7},
};
```

And for the feather m0:
```
const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {3, 6, LMIC_UNUSED_PIN},
};
```

sbalula, 2017-10-27:
Some Modules have a bad crystal. Have you tried
~~~~
LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
~~~~
It worked for me.
Best

amarion35, 2017-10-27:
Finally I succeeded.
I'm not sure how I did. I also made the uno worked. It looked like it was faster and more efficient with the uno and the shield. I think that my antenna on the feather was not very reliable compared to the lora shield. Some days the feather worked without problems (join, uplink and downlink) and the next one I couldn't join the network. The lora shield has better performances and reliability with his antenna however I never succeeded to get a downlink from the network. 

Here is the code I use on the feather. I also use the non-avr-printf branch and I've make the modifications to set the good frequency and radio in config.h.

```
/*******************************************************************************
   Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman

   Permission is hereby granted, free of charge, to anyone
   obtaining a copy of this document and accompanying files,
   to do whatever they want with them without any restriction,
   including, but not limited to, copying, modification and redistribution.
   NO WARRANTY OF ANY KIND IS PROVIDED.

   This example sends a valid LoRaWAN packet with payload "Hello,
   world!", using frequency and encryption settings matching those of
   the The Things Network.

   This uses OTAA (Over-the-air activation), where where a DevEUI and
   application key is configured, which are used in an over-the-air
   activation procedure where a DevAddr and session keys are
   assigned/generated for use with all further communication.

   Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
   g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
   violated by this sketch when left running for longer)!

   To use this sketch, first register your application and device with
   the things network, to set or generate an AppEUI, DevEUI and AppKey.
   Multiple devices can use the same AppEUI, but each device has its own
   DevEUI and AppKey.

   Do not forget to define the radio type correctly in config.h.

 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.
static const u1_t PROGMEM APPEUI[8] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
void os_getArtEui (u1_t* buf) {
  memcpy_P(buf, APPEUI, 8);
}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
void os_getDevEui (u1_t* buf) {
  memcpy_P(buf, DEVEUI, 8);
}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
// The key shown here is the semtech default key.
static const u1_t PROGMEM APPKEY[16] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
void os_getDevKey (u1_t* buf) {
  memcpy_P(buf, APPKEY, 16);
}

static uint8_t mydata[] = "Hello World!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
[... 115 lines cut ...]

#ifdef VCC_ENABLE
  // For Pinoccio Scout boards
  pinMode(VCC_ENABLE, OUTPUT);
  digitalWrite(VCC_ENABLE, HIGH);
  delay(1000);
#endif

  // LMIC init
  os_init();
  // Reset the MAC state. Session and pending data transfers will be discarded.
  LMIC_reset();
  LMIC_setLinkCheckMode(1);
  LMIC_setAdrMode(1);
  //LMIC_setDrTxpow(DR_SF9,20);
  //LMIC.dn2Dr = DR_SF9;

  // Start job (sending automatically starts OTAA too)
  do_send(&sendjob);
}

void loop() {
  os_runloop_once();
}
```

1993aicha, 2018-05-08:
hello @amarion35 ,
i try to work with your code because i had the same problem my kerlink getway does not see the join request :Starting
Packet queued
: EV_JOININGives 
but your code gives me :+1: millis()=1000

millis()=1500
millis()=2000
millis()=2500
 can u help me please !!

amarion35, 2018-05-09:
@1993aicha Maybe I can help you but you should give me more details. I didn't worked with my own gateway but only with a phone operator providing a lora network.

I assume you are trying to join a lora module on an arduino to your gateway? I also suppose that the module and the gateway are really close.

From my experience the first thing is to be sure that the module works in emission and reception and the signal is powerful enough.

genoinosonda, 2018-06-14:
i have a same problem.

Packet queued
4387819: EV_JOINING

matthijskooijman, 2020-01-26:
Since this is not really a bug, I'm closing this issue. I also added some info about downlink errors at https://github.com/matthijskooijman/arduino-lmic#problems-with-downlink-and-otaa

closed 2020-01-26 by matthijskooijman