matthijskooijman/arduino-lmic #126: How can I see downlink data?

nuguri618, 2017-10-30:
Hello!
I need to know where get the downlink data..

    if (ev == EV_TXCOMPLETE) {
    dataSent = true;
   
   if (LMIC.dataLen) {
        // data received in rx slot after tx
        Serial.print(F("Received "));
        Serial.print(LMIC.dataLen);
        Serial.print(F(" bytes of payload: 0x"));
        for (int i = 0; i < LMIC.dataLen; i++) {
          if (LMIC.frame[LMIC.dataBeg + i] < 0x10) {
            Serial.print(F("0"));
        }
        Serial.print(LMIC.frame[LMIC.dataBeg + i], HEX);
    }
    Serial.println();
I put this code for check receiving data.. but can't see any data from LoRaWan network server.
I use network server from (https://github.com/gotthardp/lorawan-server) this.
But can't see any receive data. even ACK.

matthijskooijman, 2017-10-30:
Serial.print(LMIC.frame[LMIC.dataBeg + i], HEX);

This line should be inside the for loop. Did you copy-paste this code from somewhere? It looks familiar, I'm wondering if the orignal was also wrong?

nuguri618, 2017-10-30:
Yes I copy from other one's code..and I never fix anything.

matthijskooijman, 2017-10-30:
Could you tell me *where* you copied this code from?

nuguri618, 2017-10-30:
https://www.thethingsnetwork.org/forum/t/downlink-to-node-with-lmic/5127/9
I copied from (https://www.thethingsnetwork.org/forum/t/downlink-to-node-with-lmic/5127/9)this page

matthijskooijman, 2017-10-30:
Oh, looking at your code again, it seems your code is already correct (at least the problem I noted before is not present), but I got confused because indentation is messed up. You also seem to be missing a close `}` but you probably just didn't copy that from your code.

nuguri618, 2017-10-30:
I'm not missing '}' to my code. In my case Always ( LMIC.dataLen = 0) <-like this. So I don't know where received data. 
Or I just missed setup at lorawan server? 
In Lora network server just disappeared downlink data when uploading finish. So I confuse which I made mistake..

matthijskooijman, 2017-10-30:
If dataLen is 0, then LMIC did not receive any downlink data. So either it's not being sent, or it is not being received. To distinguish, you can check the gateway logs if you have access to them. If reception does not work, you might need to relax timings (using setMaxClockErrorRate or some similar function).

PhatHub, 2017-11-08:
@matthijskooijman Maybe nurugur's having a modulation parameter issue? I had a problem where I couldn't see the downlinks but I tweaked the spread factor in my rps and it started working...

PhatHub, 2017-11-21:
A few more thoughts:

- If you have physical access to your gateway, connect it to a hub and run wireshark on the network to look for the UDP packets with JSON that starts with "txpk:". That JSON should outline the radio parameters for your downlink.
- Make sure that the LoraWan server's frame counters and the Arduino's frame counters are in sync. If the uplink frame counter is bad, then the lorawan server won't (shouldn't!) send back a PULL_ACK and PULL_RESP response to your gateway. If your downlink counter is too far out of sync, your end point should be rejecting the downlink packet.

nuguri618, 2017-11-24:
JSON down: {"txpk":{"powe":20,"imme":false,"tmst":80891260,"codr":"4/5","datr":"SF12BW500","freq":923.3,"modu":"LORA","rfch":0,"ipol":true,"size":30,"data":"oAEA/wOB6gIGASH1tglH9bCTSxzm799XJuKYsnb7"}}
INFO: tx_start_delay=1497 (1497.000000) - (1497, bw_delay=0.000000, notch_delay=0.000000)
a6.d3.33.4.d2.47.a3.b.0.9c.1e.12.0.8.0.0.a0.1.0.ff.3.81.ea.2.6.1.21.f5.b6.9.47.f5.b0.93.4b.1c.e6.ef.df.57.26.e2.98.b2.76.fb.end
lgw_receive:1179: [2 17]
Note: LoRa packet
INFO: [down] PULL_ACK received in 1 ms
lgw_receive:1179: [2 17]
Note: LoRa packet

I got this message from LoRa Gateway..
Is it success downlink work to lora mote? 
But still have downlink_lost message

PhatHub, 2017-11-27:
Well, that packet looks good. That would be from your Loraserver/LoraGatewayBridge to your gateway.
I'm inferring from your packet that the Frame counter is 746 and the DevAddr is 0x03ff0001. If you reset your Arduino device, make sure you also reset the downlink frame counter on the server side as well.

So basically, your gateway _should_ be transmitting. If you want to go one step further, you can use GQRX and an RTL USB stick to read at 923.3 MHZ to see if the downlink burst gets transmitted.