matthijskooijman/arduino-lmic #130: Question about EEPROM layout?

PhatHub, 2017-11-17:
Is there a map of what's expected to be in the EEPROM storage, if it's used?
Is there a safe offset in the EEPROM where we can store things between power cycles, like:

- DevAddr, DevEUI 
- NwkSKey, AppSKey
- Down/Up frame counters

...or any other application-specific data that we want to keep in non-volatile memory?

Thanks!

matthijskooijman, 2017-11-17:
LMIC does not do anything with EEPROM, that is left up to the application. The examples just hardcode the addresses and keys in the source code, you could load them from flash (e.g. like [this sketch](https://github.com/matthijskooijman/arduino-lmic/issues/130) for OTAA).

As for up/down frame counters, I haven't really tested with saving them. You might need to save more stuff (but you can access the frame counters through the LMIC struct, I suspect writing to the framecounters on startup should work).

PhatHub, 2017-11-20:
Excellent! I thought that there was something that got grabbed from EEPROM, but I misread the variables that had PROGMEM on it.

I guess setting up using EEPROM isn't universal, for instances like the Feather M0 where it doesn't have non-volatile storage. So it makes sense to have it populated by the end user.

closed 2017-11-20 by PhatHub