matthijskooijman/arduino-lmic #135: Stuck after TXMODE

parhamdb, 2018-01-17:
I'm using lorawan on 815Mhz on SX1276, Arduino from ESP32 on a heltec board.
I use ABP set session and start sending packets.
After random number of packets (anywhere from 2-3 to 20), the system stops after TXMODE line.
No error or anything else after that.
I tried using a timeout, resetting with LMIC_reset(), even shutting down and bringing radio back, but I can never get LoraWan started and sending packets again unless I reset ESP32 through Reset Button.

54981910: engineUpdate, opmode=0x808
57218879: Uplink data pending
58337363: Airtime available at 56100397 (channel duty limit)
59455962: Ready for uplink
60574676: TXMODE, freq=902300000, len=86, SF=10, BW=125, CR=4/5, IH=0

matthijskooijman, 2018-01-18:
Perhaps some DIO line is not solidly connected? IIRC this behaviour can be seen when the TX complete interrupt is not visible through a DIO line.

parhamdb, 2018-01-18:
Ok I did further investigation, added:

if (dio_states[i]){
                Serial.printf("hal_io_check Firing %d\r\n",i);
                radio_irq_handler(i);
            }
Line 69 of hal.cpp
So I checked if the IRQ is getting fired from lmic. For sure It wasn't getting fired and process was stuck there.
I added ESP32 code to print when interrupt is fired.
So ESP detects the interrupts and fires but for some reason LMIC is not detecting the interrupt and keeps waiting for it.
This is pretty random, This happened on first packet or 60th packet in my tests.
I tested with two different boards of the same make/model. also tried another make/model. It is pretty ESP hardware/wiring independent.

parhamdb, 2018-01-20:
I investigated this further.
Since ESP32 is a multi-core devices, apparently there can be a race condition if sending data is performed from another thread and possibly processor.
I just made sure all data sending is performed from same thread/core running arduino and LMIC and everything seems fine.
I'm closing the case.
Thanks

closed 2018-01-20 by parhamdb