matthijskooijman/arduino-lmic #136: LMiC not processing MAC commands

pansofik, 2018-01-29:
Does LMiC v1.5 process MAC commands?   I am using the stack against Actility NS and there is no processing of the downlink MAC commands that I see.   Not sure if its a HAL or radio issue.   lmic.c has the code to process MAC commands which looks correct.    I do note that the readme states MAC processing has not been tested.    Has this changed?   Obviously ADR cannot work with this.

cyberman54, 2018-01-29:
I am using LMiC 1.5 on TTN and see ADR working on my nodes, so i assume MAC commands are executed.

matthijskooijman, 2018-01-29:
Could it be downlink isn't working properly due to timing inaccuracies?

pansofik, 2018-01-29:
Could be timing.   I will check that.  Thanks for the prompt response!

cyberman54, 2018-01-30:
try a downlink with ABP join. If this does not work, you found the reason why OTAA can't work.

cyberman54, 2018-02-15:
I checked this again with LMIC 1.5 ardunio-2 and TheThingsNetworks joined via OTAA. I can confirm that ADR setting works, the node's spreasfactor is remotely controled by the network. So, MAC commands work.

pansofik, 2018-02-15:
Thanks.  I keep getting rxtimeout so maybe something with timing or how i
access the radio.   I am using a Modtronix inair9 and a 3.3v 8mhz arduino
mini pro.    I have SPI connected to correct pins and set up the dio pins
as follows:

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {2, 5, LMIC_UNUSED_PIN},
};

I am also using a clock offset (this is definitely needed as OTAA will not
work unless I use the offset:

// Deal with clock inaccuracies affecting Rx Window timing (NEED THIS for
MINI PRO!)
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);

Any ideas welcome.

Thanks

pansofik, 2018-02-15:
Did not send to all first time

Thanks.  I keep getting rxtimeout so maybe something with timing or how i
access the radio.   I am using a Modtronix inair9 and a 3.3v 8mhz arduino
mini pro.    I have SPI connected to correct pins and set up the dio pins
as follows:

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {2, 5, LMIC_UNUSED_PIN},
};

I am also using a clock offset (this is definitely needed as OTAA will not
work unless I use the offset:

// Deal with clock inaccuracies affecting Rx Window timing (NEED THIS for
MINI PRO!)
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);

Any ideas welcome.

Thanks