matthijskooijman/arduino-lmic #137: LoRaWan 1.1 DeviceTimeReq supported?

Jeroen88, 2018-01-30:
I would like to synchronize my microcontroller with UTC time. Can I use the library for this, using the LoRaWan 1.1 feature DeviceTimeReq?

cyberman54, 2018-01-30:
It won't be possible to sync a node's RTC with DeviceTimeReq when this shall happen with high precision. As far as i understand DeviceTimeReq in LoRaWAN 1.1 this command does not take care of latency and jitter of the LoRaWAN connection. So additional timr sync logic will be required to set a clock with precision of e.g. +/- 500ms

Jeroen88, 2018-01-30:
@cyberman54 thnx for your quick response. 500 ms would be great. My microcontroller can keep clock time but must be set / synched regularly. 

Any chance of support of this feature in the near future? Any example of how to get the DeviceTimeReq myself making maximum use of the abstraction of the library?

VladimirAkopyan, 2019-03-16:
Also interested, I just need rough tune fur calendar keeping. +/- 1 minute will do for me