matthijskooijman/arduino-lmic #141: C:\Users\Vivek\Documents\Arduino\libraries\Time-master\libraries\arduino-lmic-master\src\lmic\radio.c:689

designelektronik, 2018-02-07:
Hi,

Few minutes back, i could able to receive message in my server and about 45 message i had received. But after sometime, the terminal showed this failure status and stopped transmitting the data. 
Again i hardcoded the program and opened the terminal, it is showing the failure status.
Why this is happening and how to solve this issue.

i had custom designed board and the pin mapping is here,

const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 8,
    .dio = {2, 6, 7},
};

help me to resolve this issue asap.

thanks
saikumar

matthijskooijman, 2018-02-07:
By e-mail, @designelektronik added that the assert in question is: 

    ASSERT(v == 0x12 );

This means that the version number read from the transceiver does not match what was expected. In practice, this usually means that SPI communication is completely failing and it is reading only zeroes. Given that communication worked initially, I would suspect a loose wire or solder contact somewhere in the SPI lines.

designelektronik, 2018-02-07:
Hi,

Thanks for your reply. I cross checked with the soldering. The connectivity found to be perfect . But still I face this issue. Is there any alternate way to resolve this issue.

Also would like to know, is there any limitation on medsage transmitted by the transceiver?
Because on every 110th message my transmitter stops transmitting the data when I use the dragino gps shield+ arduino uno stack.

Kindly help me to resolve 

Thanks

matthijskooijman, 2018-02-07:
> Thanks for your reply. I cross checked with the soldering. The connectivity found to be perfect . But still I face this issue. Is there any alternate way to resolve this issue.

You could modify the library code to print the "v" variable, that might provide a hint about what's going on (but I expect it is either 0xff or 0x00, both suggest a problem in the SPI communication). Having a logic analyzer on the SPI pins might be helpful, but I won't be able to do hand-holding through that debugging process...

> Also would like to know, is there any limitation on medsage transmitted by the transceiver?
Because on every 110th message my transmitter stops transmitting the data when I use the dragino gps shield+ arduino uno stack.

Nope, there are no (practical) limits on the number of messages. If things stop sending, I would suspect a memory look or other logic error in your code.

designelektronik, 2018-02-26:
Hi,

Could you please let me know how can we modify the hal.cpp & hal.h incase if the spi module has TxEN & RXEN Pins.

I followed the same procedure and the module struck at the packet queued, the Tx window has not opened for the transmission.

i followed the modifications done by Nicolaivr/arduino-lmic and it didnt work out

guide me to resolve this tx window issue

Thanks

[image]

madc, 2018-04-30:
We've seen the same issue. The reason was a missing connection (LoRaWAN module had no power).

Abinjoy1999, 2020-03-10:
[image]
what will be the error