matthijskooijman/arduino-lmic #144: got 868 working, cant get 915 working

lagunacomputer, 2018-02-19:
Hello i was able to get 868 working with this LMIC code for the node, and things4u/ESP-1ch-Gateway-v4.0 or v5.

so i bought some 915 mhz modules.
Does anyone have a working 915 code set for ESP8266?
I got the gateway up, and it can receive packets on 903.9, however i never get TNN to show 'Gateway status connected' on the web dashboard.

More frustrating, im not sure how to correctly pick the 903.9 channel on the transmitting node using LMIC code.  Any one with a working copy of this in 915 mode, please post or upload i would be very grateful , thanks

lagunacomputer, 2018-02-19:
i finally figured this out, I had to delete my existing TNN gateway and app, and recreate them

graypickney, 2018-05-01:
Was your gateway originally configured for 915 mhz and you had to recreate it anyway?

I commented the Euro frequency in the config file and uncommented 915 mhz. I've been able to connect to my Gateway that's setup for 915, but oddly enough, the uplink messages are 868 mhz while the downlink is 915, as it should be. 

Could anyone help me out here?

lagunacomputer, 2018-05-01:
I was able to get this working on 915, perhaps I can zip up the code and
post it somewhere for you (at work right now)

lagunacomputer, 2018-05-01:
"Was your gateway originally configured for 915 mhz and you had to
recreate it anyway?"

yes, I believe I just had to delete the TNN gateway device and recreate it,
without changing the code

graypickney, 2018-05-02:
That'd be great, thank you. Did you have to make any significant changes other than:

1. Update config.h to 915 mhz
2. Change in hal.cpp
           SPISettings settings(10E6, MSBFIRST, SPI_MODE0); 
    to  
           SPISettings settings(8E6, MSBFIRST, SPI_MODE0);

3. Add clock fix line of code: LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);

The weird thing is I'm able to get an EV_TXCOMPLETE, but when I look at the Gateway traffic, the uplinks are all at 868 mhz in the payload, so I'm not sure how they're reaching the 915mhz gateway anyway.

graypickney, 2018-05-02:
Okay, I'll see if I can restart the gateway. Here's hoping.

lagunacomputer, 2018-05-02:
obviously, you will need to change all your IDs. keys, ssid, wifipassword etc.  Here is working code for both 868 and 915, node and gateways.  I use NodeMCU ESP8266's and sx1276 modules
[Lora_node_esp8266_ttn-abp_915.zip](https://github.com/matthijskooijman/arduino-lmic/files/1965604/Lora_node_esp8266_ttn-abp_915.zip)
[lora-ESP8266-sc-gway_915_WORKING.zip](https://github.com/matthijskooijman/arduino-lmic/files/1965605/lora-ESP8266-sc-gway_915_WORKING.zip)
[lora-ESP8266-sc-gway_915.zip](https://github.com/matthijskooijman/arduino-lmic/files/1965606/lora-ESP8266-sc-gway_915.zip)
[lora-ESP8266-sc-gway_working_868_final.zip](https://github.com/matthijskooijman/arduino-lmic/files/1965607/lora-ESP8266-sc-gway_working_868_final.zip)
[Lora_node_esp8266_working_ttn-abp_868_final.zip](https://github.com/matthijskooijman/arduino-lmic/files/1965608/Lora_node_esp8266_working_ttn-abp_868_final.zip)

lagunacomputer, 2018-05-02:
hi I don't remember doing #2 or #3

graypickney, 2018-05-02:
Thanks for the files, I took a look. I see you're using ABP authentication. I've gone with the OTAA route, so its not quite the same, though we do have the same change in config.h to disable 868 mhz configurations and enable 915 mhz.

Reinstalled the gateway and device on ttn-console:

I'm still able to start the node, join, and see a success message, however the gateway still shows joins and uplinks as 868 mhz. I don't see how this is possible as the gateway is configured to use ttn-router-us-west, and is on 915mhz frequency plan. Can anyone shed some light on how/why this may be happening? It shouldn't even be able to send messages at 868 mhz, as my antenna is configured for 915 mhz as well... so confused. Perhaps I'm reading the console data incorrectly?

lagunacomputer, 2018-05-02:
loramodem.h ?

use all the files in the zips they are all edited

closed 2018-08-19 by lagunacomputer