matthijskooijman/arduino-lmic #152: AES error with STM32 Core Arduino branch and non-avr-printf

sabas1080, 2018-03-18:
Hi

I'm trying to use with STM31L1, but when trying to compile I get the following error in Arduino IDE 1.8.4

`/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/aes.c.o: warning: multiple common of `AESAUX'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o: warning: previous common is here
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/aes.c.o: warning: multiple common of `AESKEY'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o: warning: previous common is here
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/aes.c.o: In function `os_aes':
aes.c:(.text.os_aes+0x0): multiple definition of `os_aes'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o:lmic.c:(.text.os_aes+0x0): first defined here
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o: In function `aesroundkeys':
lmic.c:(.text.aesroundkeys+0x72): undefined reference to `table_get_u1'
lmic.c:(.text.aesroundkeys+0x84): undefined reference to `table_get_u1'
lmic.c:(.text.aesroundkeys+0x96): undefined reference to `table_get_u1'
lmic.c:(.text.aesroundkeys+0xa8): undefined reference to `table_get_u1'
lmic.c:(.text.aesroundkeys+0xc0): undefined reference to `table_get_u4'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o: In function `os_aes':
lmic.c:(.text.os_aes+0x23e): undefined reference to `table_get_u4'
lmic.c:(.text.os_aes+0x254): undefined reference to `table_get_u4'
lmic.c:(.text.os_aes+0x26a): undefined reference to `table_get_u4'
lmic.c:(.text.os_aes+0x27e): undefined reference to `table_get_u4'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o:lmic.c:(.text.os_aes+0x292): more undefined references to `table_get_u4' follow
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o: In function `os_aes':
lmic.c:(.text.os_aes+0x690): undefined reference to `table_get_u1'
lmic.c:(.text.os_aes+0x6a8): undefined reference to `table_get_u1'
lmic.c:(.text.os_aes+0x6c0): undefined reference to `table_get_u1'
lmic.c:(.text.os_aes+0x6d6): undefined reference to `table_get_u1'
lmic.c:(.text.os_aes+0x6f2): undefined reference to `table_get_u1'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/aes/lmic.c.o:lmic.c:(.text.os_aes+0x70a): more undefined references to `table_get_u1' follow
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/lmic.c.o: In function `calcRxWindow':
lmic.c:(.text.calcRxWindow+0x72): undefined reference to `table_get_ostime'
lmic.c:(.text.calcRxWindow+0x98): undefined reference to `table_get_ostime'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/lmic.c.o: In function `calcBcnRxWindowFromMillis':
lmic.c:(.text.calcBcnRxWindowFromMillis+0x46): undefined reference to `table_get_ostime'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/lmic.c.o: In function `schedRx2':
lmic.c:(.text.schedRx2+0x20): undefined reference to `table_get_ostime'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/lmic.c.o: In function `txDone':
lmic.c:(.text.txDone+0x9c): undefined reference to `table_get_ostime'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/lmic.c.o: In function `processDnData':
lmic.c:(.text.processDnData+0x60): undefined reference to `table_get_u1'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/radio.c.o: In function `rxlora':
radio.c:(.text.rxlora+0x9c): undefined reference to `table_get_u1'
/var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_77258/libraries/arduino-lmic/lmic/radio.c.o: In function `radio_irq_handler':
radio.c:(.text.radio_irq_handler+0x66): undefined reference to `table_get_u2'
collect2: error: ld returned 1 exit status`

Thanks

thebenst, 2019-09-20:
Hello sabas1080,
can you please describe the solution in more detail?

Tobias

antoineQue, 2019-09-23:
Hello,

Can you post a detailed explanation, please.

Thanks.

sabas1080, 2019-09-24:
In arduino IDE with STM32duino

[image]

thebenst, 2019-09-24:
great, thank you very much!