matthijskooijman/arduino-lmic #153: OTAA observations

ticso24, 2018-03-20:
I've read several mentions about OTAA joining problems.
I have had the same problem.
It took a very long time to join, sometime half an hour or even more.
I saw that my gateway delivered the join request to TTN and I assume also the join answer was send by my gateway (Raspberry/RAK831 based), at least it got UDP packets from TTN.
Long story short: the cause was hardware related on the node
I've used a chinese Arduino Mega clone with a Dragino Shield and changed to a chinese Uno clone with another Dragino shield and OTAA worked extremly well with the Uno.
To rule out the Shield I've put the shield, which I'd used on the Mega, and put it on the Uno and the Uno still worked, so something on the Mega board causes the issue.
There is not very much which could go wrong with the AVR I assume, beside the quality of the xtal oscillator and maybe power quality.

albtrentadue, 2018-09-17:
Hi ticso24
I try to bring up this topic, I have similar issue about OTAA join.
I am using Arduino PRO Mini running the LMIC, powered using an FTDI USB-to-Serial adapter and driving a HopeRF RFM96H LoRa radio chip.
We see that the join fails several times (4-10 times), then eventually succeeds and then subsequent sensor data transmissions are always successful.
I am rather sure (by looking into the logs) that the Network Server (Loraserver.io) handles the join request and responds to it, and the packet gateway relays all packets with no error.
May this issue be brought down to the node HW? 
I may agree that the RFM96H + Arduino Pro Mini is a (very) low end combination, but why data transmission never fails after the first successful OTAA join? We also made longer distance data transmissions test after join and they do not fail, so I am not sure power supply or HW is the root cause.
Any suggestion?
Thanks in advance.

cyberman54, 2018-09-17:
Sounds like RX problem, possibly a timing issue.
RX is needed with OTAA only.
To debug this, active LMIC debug level and carefully analyse the timestamps during OTAA phase.

petzah, 2018-10-05:
@albtrentadue

try to add
    `LMIC_setClockError(MAX_CLOCK_ERROR * 2 / 100);`

after LMIC_reset()

see https://github.com/matthijskooijman/arduino-lmic/issues/22