matthijskooijman/arduino-lmic #155: FAILURE  C:\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:523

gpiscopo, 2018-03-21:
Hi,
I use the board Adafruit Feather M0 with LoRa module RFM95 (868MHz). I am trying to send a string to another feather and to a gateway. I am using the LMIC library by @matthijskooijman and I did these modifications:
-In the HAL.cpp file: `static const SPISettings settings(8E6, MSBFIRST, SPI_MODE0);`
-In the config.h file: `#define CFC_eu868` and `#define CFG_sx1276`
-In the setup function: `LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);`

The pinout I use is:
```
// Pin mapping found on Internet according to the wires that I soldered
const lmic_pinmap lmic_pins = {
.nss = 8, // change to match cs / nss
.rxtx = LMIC_UNUSED_PIN,
.rst = 4, // possible 4 for the adafruit board
.dio = {3, 6, 11}, // DIO0, DIO1, DIO2 data pins
};
```

My problem arises when I try to send or receive something to or from the other feather board and I get this error: FAILURE  C:\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:523
[image]

If I implement the communication with the other feather only, my code works for sure. The same if I communicate only with the gateway. I see the error only if I try to implement both the communications with the other feather and the gateway.

These are the RFM95 settings I am using:
```
/* for feather m0 */
#define RFM95_CS 8
#define RFM95_RST 4
#define RFM95_INT 3

// Change to 434.0 or other frequency, must match RX's freq!
#define RF95_FREQ 868.0
 
// Singleton instance of the radio driver
RH_RF95 rf95(RFM95_CS, RFM95_INT);
```

Any ideas on how to solve this problem (if feasible)?

Thank you very much!

nafitriaulia, 2018-04-30:
Hi 
I got same problem with you. Do you already know how to fix it?

gpiscopo, 2018-04-30:
Hi,
I found that you can implement only one of two communication per time

nafitriaulia, 2018-05-08:
Do you ever try communicate node (Arduino + LoRa) to gateway (Arduino + Lora) and successfully? using LMIC library?

gpiscopo, 2018-05-08:
Hi,

I am able to have a communication between to feather m0 equipped with Lora modules, or a communication between many feather m0 equipped with Lora modules and a gateway. It is worth noting that in case you want to communicate with a gateway you cannot have anymore the communication between the feathers, only between the feathers and the gateway.

Roberto6969, 2020-02-10:
Long story short: i have node, consisted of Mega2560 Core and RFM95. Works fine, but after about half of day after start I getting error:
- 17:47:08.457 -> Packet queued
- 17:47:12.372 -> FAILURE 
- 17:47:12.372 -> C:\Users\My_name\Dropbox\Arduino\src\libraries\arduino-lmic-master\src\lmic\radio.c:523

What is the cause of this error?