matthijskooijman/arduino-lmic #156: Failure at Arduino\libraries\arduino-lmic-master\src\lmic\lmic.c:1889

Gabi12345678, 2018-03-24:
Hi, 

I am new to Arduino and am trying to use the LMIC library for a university project. I started by trying the example code: examples/ttn-abp/ttn-abp.ino. I am currently using an Adafruit Feather RFM95 LoRa Radio - 900MHz - RadioFruit. After I upload the program, the board manages to connect to the gateway. The program triggers the onEvent function and the console prints the "EV_TXCOMPLETE (includes waiting for RX windows)" output. However, after a couple of seconds where I assume it is trying to send the message a second time, I receive FAILURE at Arduino\libraries\arduino-lmic-master\src\lmic\lmic.c:1889. Could anybody please explain what the cause of this might be and how I might be able to solve it?

My pin assignments are:
`
const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {3, 6, LMIC_UNUSED_PIN},
};
`
Thank you very much in advance.

nafitriaulia, 2018-04-30:
Hello

i got same problem with you. Do you already know how to fix it?

Gabi12345678, 2018-04-30:
Hi,

For me, it was a combination of two things that were going wrong. I recommend you read this thread 

https://www.thethingsnetwork.org/forum/t/lmic-on-adafruit-lora-feather-successfully-sends-message-to-ttn-and-then-halts-with-packet-queued/3762

 which highlights some problems in both Adafruit Feather versions. The second problem was that my Feather M0 was faulty. I think there was something either at the clock or the antena which caused that ASSERT to trigger. I didn't investigate further I just bought a new chip which immediately worked.

Hope this helps