matthijskooijman/arduino-lmic #157: Not recieving Downlink messages from gateway

arihantdaga, 2018-03-26:
I am using Dorji DRF1276G Module Coupled with Atmega 328. And I am using loraserver.io on my server. So far, sending data from the node is working great. I am able to send the data from the node to the gateway which is sending to my server as well. 
But none of the messages i send from gateway to the device are recieved here - in short these lines are never executed - 

```
Serial.println(F("Received "));
Serial.println(LMIC.dataLen);
Serial.println(F(" bytes of payload"));
```

Does This library support downlink ? Or am i doing something wrong, can anybody help me out here..

cyberman54, 2018-03-26:
yes, it supports downlinks and it works. Check your setup, i.e. cabling of DIO lines of SX127x chip. DIO0 and DIO1 *must* be wired to GPIO ports of the CPU where lmic is running. Pin mappings in configfile must be set accordingly to wiring.

arihantdaga, 2018-03-27:
@cyberman54 Thank you very much. Its working now. In my case i had to set this in code 

` LMIC.dn2Dr = DR_SF9;`

as mentioned in the readme. 
Although i am missing it sometimes.. infact it worked fine initially and now its not working again.. 
I enabled debug and set LMIC_DEBUG_LEVEL to 1
After that this is what i am getting on serial window. 

```
Packet queued
9559710: RXMODE_SINGLE, freq=867500000, SF=7, BW=125, CR=4/5, IH=0
9563060: Received downlink, window=RX1, port=-1, ack=0
9563110: EV_TXCOMPLETE (includes waiting for RX windows)
9563325: engineUpdate, opmode=0x800
```
 While this is what i have in my ttn.ino -

```
case EV_TXCOMPLETE:
           Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
           if (LMIC.txrxFlags & TXRX_ACK)
             Serial.println(F("Received ack"));
           if (LMIC.dataLen) {
                Serial.println(F("Received "));
                Serial.println(LMIC.dataLen);
                Serial.println(F(" bytes of payload"));
                Serial.print("txCnt :"); Serial.println(LMIC.txCnt);
                Serial.print("txrxFlags :"); Serial.println(LMIC.txrxFlags);
                Serial.print("dataBeg :"); Serial.println(LMIC.dataBeg);
                for (int i = 0; i < LMIC.dataLen; i++) {
                    if (LMIC.frame[LMIC.dataBeg + i] < 0x10) {
                    Serial.print(F("0"));
                    }
                    Serial.print(LMIC.frame[LMIC.dataBeg + i], HEX);
                }
           }
           // Schedule next transmission
           os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
           break;
```

arihantdaga, 2018-03-27:
And sometimes i am getting this in serial logs

```
Packet queued
288414458: RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=0
288477694: RXMODE_SINGLE, freq=866550000, SF=10, BW=125, CR=4/5, IH=0
288480158: EV_TXCOMPLETE (includes waiting for RX windows)
288480212: engineUpdate, opmode=0x900
```

AI2M, 2018-07-06:
can i see yours code that working?. i try to solve it but i can't. 

thank you.

Shashank545, 2018-08-29:
I have implemented the single channel packet forwarder LORA gateway and it is able to send data from node to server through gateway. But I am not able to send data from gateway to device to control led. i a m also able to send downlink data from TTN to gateway and gateway is also able to send it . My led actuation is not happening. Here is what I have used.

Lora Node : Arduino UNO + Dragino LoRa Shield
Lora Gateway : Rasperry Pi LoRa HAT + RaspberryPi3B

Any immediate and urgent help would be appreciated.

sprokkie, 2019-10-13:
have same problems here, no data recieved bij de heltec 32 module.

in the serial log:

Packet queued
20246753: RXMODE_SINGLE, freq=867700000, SF=7, BW=125, CR=4/5, IH=0
20306501: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
20320093: EV_TXCOMPLETE (includes waiting for RX windows)
20322820: engineUpdate, opmode=0x900
22197820: engineUpdate, opmode=0x908
22197867: TXMODE, freq=867900000, len=26, SF=7, BW=125, CR=4/5, IH=0
Packet queued
22261225: RXMODE_SINGLE, freq=867900000, SF=7, BW=125, CR=4/5, IH=0
22320973: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
22334565: EV_TXCOMPLETE (includes waiting for RX windows)
22337292: engineUpdate, opmode=0x900

the ino file the samen as above 

`case EV_TXCOMPLETE:

           Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
           if (LMIC.txrxFlags & TXRX_ACK)
             Serial.println(F("Received ack"));
           if (LMIC.dataLen) {
                Serial.println(F("Received "));
                Serial.println(LMIC.dataLen);
                Serial.println(F(" bytes of payload"));
                Serial.print("txCnt :"); Serial.println(LMIC.txCnt);
                Serial.print("txrxFlags :"); Serial.println(LMIC.txrxFlags);
                Serial.print("dataBeg :"); Serial.println(LMIC.dataBeg);
                for (int i = 0; i < LMIC.dataLen; i++) {
                    if (LMIC.frame[LMIC.dataBeg + i] < 0x10) {
                    Serial.print(F("0"));
                    }
                    Serial.print(LMIC.frame[LMIC.dataBeg + i], HEX);
                }
           }
           // Schedule next transmission
           os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
           break; `

any ideas what goes wrong ?

@arihantdaga What code do you use ?

sprokkie, 2019-10-14:
i use the heltec esp32 lora module, it works well only no message from the thingsnetwork to the node.
any one else who can help ?