matthijskooijman/arduino-lmic #158: [help] os_runloop hangs code execution

paulmassen, 2018-03-30:
Hi, 

Sorry if this will be a basic question, but I spent several hours on this and could not find a solution. 

Basically, I use the commen Heltec lora board and want to send to TTN whenever a RFID read happens.
I managed to send basic command on rfid read as seen here: https://github.com/paulmassen/heltec-wifi-lora/blob/master/workingrfidandlora

I too managed to send basic message to my TTN app with your example code, but I can't mix them together. 

I tracked the problem down to os_runloop which seems to hang the execution of the code which is written after.

https://github.com/paulmassen/heltec-wifi-lora/blob/master/draft.ino#L73

Am I missing something really obvious here? Is there somewhere some examples of lmic being used with sensors? I could not find any.

Many thanks for your help.

matthijskooijman, 2018-03-30:
This is exactly what `os_runloop` is supposed to do: Infinite loop and run scheduled tasks. If you want to run just a single loop, use `os_runloop_once` (which you need to call very often to make sure tasks can run at the right time, but you can do other things in between).

paulmassen, 2018-03-30:
ohh, ok, thanks a lot!

Then, would you have some insights if I would want to trigger a lora message on a given event?
I'm currently trying to wait for my event, and when event is triggered, it runs SPI.begin with Lora (my rfid reader is set with spi too) but I have a FAILURE message which mentions radio.c line 473 which is:

`ASSERT((readReg(RegOpMode) & OPMODE_LORA) != 0);`

Any idea?

Thanks again