matthijskooijman/arduino-lmic #162: LoRa32u4 II never gets EV_TXCOMPLETE

j4zzcat, 2018-04-15:
Hi,

I'm testing with the LoRa32u4 II board. The node joins fine to the TTN network, sends the first payload, but then never sends again, as EV_TXCOMPLETE is never called.

j4zzcat, 2018-04-15:
Problem solved. I had to solder DIO1 to D1. For reference, DIO1 is located on the back side of the board. See screen shot.

Here’s the LMIC pin mapping:
```
const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 4,
    .dio = {7, 1, 2},
};
```
[image]

closed 2018-04-15 by j4zzcat

KonstantinMastak, 2018-06-28:
j4zzcat, I have the same problem with this exact board. I connected DIO1 and D1, my log changed a bit, but the problem is still here - server get my join request and sends join accept, but  LoRa32u4 II  seems to never receive it.

Here is what I get:

Starting
RXMODE_RSSI
312940: engineUpdate, opmode=0x8
313046: Scheduled job 0x26f, cb 0x2323 ASAP
Packet queued
313190: Running job 0x26f, cb 0x2323, deadline 0
313321: EV_JOINING
313376: engineUpdate, opmode=0xc
313465: Uplink join pending
313541: Airtime available at 604450 (previously determined)
313697: Uplink delayed until 604450
313804: Scheduled job 0x26f, cb 0x4b0 at 604325
314613: irq: dio: 0x2 flags: 0x0
314716: Cleared job 0x26f
314792: Scheduled job 0x26f, cb 0x4b0 ASAP
314910: Running job 0x26f, cb 0x4b0, deadline 0
315038: engineUpdate, opmode=0xc
315127: Uplink join pending
315202: Airtime available at 604450 (previously determined)
315359: Uplink delayed until 604450
315465: Scheduled job 0x26f, cb 0x4b0 at 604325
315879: irq: dio: 0x2 flags: 0x0
315983: Cleared job 0x26f
316059: Scheduled job 0x26f, cb 0x4b0 ASAP
316177: Running job 0x26f, cb 0x4b0, deadline 0
316305: engineUpdate, opmode=0xc
316394: Uplink join pending
316470: Airtime available at 604450 (previously determined)
316626: Uplink delayed until 604450
316733: Scheduled job 0x26f, cb 0x4b0 at 604325
317159: irq: dio: 0x2 flags: 0x0
317261: Cleared job 0x26f
317337: Scheduled job 0x26f, cb 0x4b0 ASAP
317455: Running job 0x26f, cb 0x4b0, deadline 0
317583: engineUpdate, opmode=0xc
317673: Uplink join pending
317748: Airtime available at 604450 (previously determined)
317904: Uplink delayed until 604450
318011: Scheduled job 0x26f, cb 0x4b0 at 604325
318418: irq: dio: 0x2 flags: 0x0
318521: Cleared job 0x26f
318598: Scheduled job 0x26f, cb 0x4b0 ASAP
318713: Running job 0x26f, cb 0x4b0, deadline 0

And so on, it repeats over and over again. 

Did you have such a problem? Did you connect or alter the board any way other than connecting DIO1 to D1? 

Thank you!

j4zzcat, 2018-06-28:
The only change was connecting DIO1 to D1. Care to paste your source code?

KonstantinMastak, 2018-06-28:
Thanks to fast reply!

I can paste it, but this is unchanged OTA sample from arduino-lmic with an
only change to pins mapping that you provided.

An update. Finally, it joined. This happened when I closed terminal monitor
in Arduino. TTN forum post tells that in v 1.2 of this board IO1 is
actually TX of Serial port. So it seems there is some conflict. Did you
observe
such problem? Do you manage to use serial monitor and working LoRa
connection at the same time?

ᐧ

j4zzcat, 2018-06-28:
I'm working straight off the command line, skipping the Arduino IDE altogether. For serial monitor I use picocom and that doesn't seem to interfere with the LoRa radio.

KonstantinMastak, 2018-06-28:
Thanks!
ᐧ

csonthomarci, 2018-11-13:
Hi guys! 

Don't you know is it working if i connect the DIO1 pin in the front of the lora32u4 module with the D6  pin via jumper cable, without any soldering?

I found the pinmapping under for solutions like this: 
const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 4,
    .dio = {7, 6 , LMIC_UNUSED_PIN}
};