matthijskooijman/arduino-lmic #164: TTGO LoRa32 V2.0 sx1278 parameters

Fbo06, 2018-04-15:
Hello, can someone please tell me how i can configure this library to work with the sx1278 ? any advice ? thank you for advance.

aplathan, 2018-04-22:
TTGO LoRa32 V2.0 works with this pinout mapping:

const lmic_pinmap lmic_pins = {
  .nss = 18, 
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 14,
  .dio = {/*dio0*/ 26, /*dio1*/ 33, /*dio2*/ 32}
};

Fbo06, 2018-04-22:
Thank you aplathan, but can you tell me please what is the other modifications you made to the code and where, because when i just replace the pinout mapping on my code i get this error message : Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:689

cyberman54, 2018-04-22:
As far as i know you need to connect DIO1 with GPIO33 by an external wire.

Fbo06, 2018-04-22:
Hello cyberman54, thank you for your response but i don't understand because in the mapping pinout my dio1 is already equal to the pin 33 and this pin is also named GPIO33 ? correct me please if i made a mistake

aplathan, 2018-04-22:
You need to connect DIO1 to GPIO33 and DIO2 to GPIO32. This is true for the V2.0 board.

Fbo06, 2018-04-23:
Hello, sorry for this late response, (->aplathan) for the config.h file, do you know which one is used (the path) because i have at least 5 of it :( ? and do i replace the standard one or do i must append the new config.h to the standard one ? for DIO1-GPIO33 and DIO2-GPIO32, i just made this modifications on my code but no hardware connections. Do i made a hardware connections ? Thank you for your patience.

skyladle, 2018-07-18:
Hi, I have same error message issue hereinafter, TTGO V2 do not see any response, how come the external wire connect or not will show the error ?

Starting
FAILURE 
D:\arduino\1.8.3\arduino-1.8.3-windows\arduino-1.8.3\hardware\espressif\esp32\libraries\arduino-lmic-master\src\lmic\radio.c:689

skyladle, 2018-07-19:
Hi ,connect DIO1 to GPIO33 and DIO2 to GPIO32 ,  but still showing 
FAILURE
D:\arduino\1.8.3\arduino-1.8.3-windows\arduino-1.8.3\hardware\espressif\esp32\libraries\arduino-lmic-master\src\lmic\radio.c:689
[image]

DeuxVis, 2018-07-19:
1) For lorawan / thethingsnetwork, dio2 is not necessary
2) You should activate arduino verbose compilation log in "File -> Preferences" : 
[image]
and show us the output.

cyberman54, 2018-07-19:
That is the part in radio.c where RegVersion is checked. It seems your LoRa chip, probably a HPD13, reports different version than expected by LMIC. Put a Log line in radio.c and check what your LoRa chip answers. Then eventually switch type between 1276/1272.

skyladle, 2018-07-20:
@cyberman54 , thanks ,what I bough is TTGO V2,, document say it is sx1278, I do not know LMIC support ?
i will try to  read  out reg version of  SX1278

trlafleur, 2018-07-20:
I believe that the SX1278 is a ~433mhz(140-512MHz) version of the LoRa
chip... should work just fine if the correct pins are wired correctly...

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

cyberman54, 2018-07-21:
Semtech chips sx1276/77/78/79 all answer with same RegVersion 0x12. 
1278 is 433 Mhz version.

Some TTGO boards do not have original Semtech chip, they use HPD13 modules (which probably use semtech chip). 

If RegVersion assert in lmic fails, this mostly happenes because the SPI interface to chip is not working. This mostly is caused by erraneous wiring or pin configuration.

So double check wiring and pin configuration, and check that you selected proper chip type (1272/1276).

soubinet, 2018-09-03:
For peoples who have FAILURE issue on radio.c:689, i have found the solution for the TTGO T-Beam T22_V07 edition. This issue is due to SPI pin configuration in "hal.cpp".

By default, the pin mapping of arduino is not matching TTGO T-Beam pinout.
The easiest way to solve the issue is to modify the "hal.cpp" file in the library of lmic (in Arduino library folder).

On the function hal_spi_init(), "SPI.begin();" should be replaced by "SPI.begin(5,19,27);". Be careful, if you use lmic library at the same time for some other devices, the pinout could be different.
For TTGO T-Beam, the pinout is :
5 = SCK
19 = MISO
27 = MOSI

I will open an issue to be able to select the SCK/MISO/MOSI from lmin init and not by modify the hal.cpp.

Hope it could help somebody.