matthijskooijman/arduino-lmic #166: High power consumption (RF96)

T94T, 2018-04-20:
Hi,

My setup is like this:
- Arduino (Barebone **ATmega328P, 3.3V, 8MHz**)
- **RF96** module

```
const lmic_pinmap lmic_pins = {
  .nss = 5,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 4,
  .dio = {3, 10, LMIC_UNUSED_PIN} 
};
```
I'm using the ttn-abp example provided in this library.

Sending data works like expected. Also the EV_TXCOMPLETE event gets triggered. But the power consumption is by far too high!
When sending there's a peak current of around 90mA. After that power consumption falls to around **40mA** and keeps staying at this level. The ATmega consumes 5mA in running mode.

It tried now three different boards with three unique RF modules. On each setup I have the exact same problem.

matthijskooijman, 2018-04-20:
The LMIC library should configure the radio into deep sleep mode whenever it is not transmitting or receiving, which works as expected in my experience. Perhaps there is something else taking up current? Did you measure the atmega without the radio attached? Did you measure the radio supply pins to see if that really is where the current goed?

T94T, 2018-04-20:
I tried it without the modules. Power consumption is as expected.

As soon as os_init(); gets called current consumption jumps up to 40mA. I'm going to check each data pin now.

T94T, 2018-04-20:
Found it! It was the pin 10 which is used on the Arduino as SS by default. It is defined as an output. So it shortens in my setup and draws 40mA.

closed 2018-04-20 by T94T