matthijskooijman/arduino-lmic #171: PEER TO PEER COMMUNICATION BETWEEN LORA NODE WITH LMIC

nafitriaulia, 2018-05-02:
I want to try lora node to talking each other with LMIC library. i am using arduino uno + Lora and Arduino Mega + Lora. How to make these two lora nodes talk to each other with raw.ino?

matthijskooijman, 2018-05-02:
Using direct communication with LMIC isn't really supported, the raw.ino example really is kept around for historical reference. If you want node-to-node communication, I suggest you look at the Radiohead library instead, which should support Lora radios as well.

closed 2018-05-02 by matthijskooijman

nafitriaulia, 2018-05-02:
I have ever tried radiohead library. it is work. but i have a question. can end node (Arduino uno + Lora) send data to Gateway (Arduino Mega + Lora) use LMIC library? 
I have project about sending data from end node to gateway (Arduino mega + Lora) using LMIC library. I have researched in last a month but i still don't get it.

I hope you want to explain me.

matthijskooijman, 2018-05-02:
Sending data to a gateway typically happens using the LoRaWAN protocol, which is what LMIC can do. However, the gateway side has typically a different radio that can listen on multiple channels and settings at the same time. It also has different software. You can fake a gateway using a normal node radio (called a "single channel gateway"), but then it only works if the node is fixed to a particular frequency and settings. I don't have any experience with single-channel gateways, though.

In any case, this issue tracker isn't really the right place for getting this kind of help, and I at least don't have time to support you more elaborately (but others might). Perhaps you should look around the TTN website and forum, which might be a better place to get help (at least on running with a (single channel) TTN gateway).

nafitriaulia, 2018-05-02:
Thank you Mr mathij for your explanation
I think i got the point and some new knowledge

Have a nice day

nafitriaulia, 2018-05-04:
Dear Mr. Matthij

I want to ask again about comment in raw.ino example..
it says "Running this sketch on two nodes should allow them to communicate. "
it means, is true if i run this sketch in two nodes then nodes can communicate each other?

thank you

matthijskooijman, 2018-05-11:
@nafitriaulia, yup, that's what it means.