matthijskooijman/arduino-lmic #172: US region configured Node still transmitting at 868 Mhz according to TTN Console

gpickney, 2018-05-02:
I have the Adafruit M0 LoRa feather.
I've cloned the latest master branch of this repo and made a few changes:

1. In 'config.h' I've commented the line for EU and uncommented #define CFG_us915 1

2. In 'hal/hal.cpp' I've changed
        static const SPISettings settings(10E6, MSBFIRST, SPI_MODE0);
        to
        static const SPISettings settings(8E6, MSBFIRST, SPI_MODE0);

3. I've added the statement 'LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);' after 'LMIC_reset();'
    in the ttn-otaa example .ino file.

Generally the logs look like this: 
EV_JOIN_FAILED (Sometimes a few times...)
EV_JOINING 
EV_JOINED
EV_TXCOMPLETE (includes waiting for RX windows)"  
Packet queued
EV_TXCOMPLETE (includes waiting for RX windows)"  
Packet queued
etc...
which I believe to be correct.

**HOWEVER**

When I view these transmissions in TTN Console, they show as 868 Mhz in the message metadata. My Gateway is 915mhz, and shows the reception of both the joins and uplinked messages, but the uplinks are at 868 mhz while the downlinks are at 915 mhz. 

What gives? Does anyone have some insight for me on this issue? Am I reading the message logs incorrectly?

matthijskooijman, 2018-05-02:
That sounds really weird, I would suspect some issue with TTN here. In particular, if your gateway is 915Mhz, it should not even detect packets on 868Mhz. Also, downlinks are supposed to be sent on the same frequency as the preceding uplink IIRC, at least for 868Mhz.

If you increase the logging level to most verbose level (3 IIRC, see config.h), I think LMIC should print the frequency it is planning to use to serial.

gpickney, 2018-05-03:
Checked the more verbose logs:
```

13227995: TXMODE, freq=907900000, len=23, SF=10, BW=125, CR=4/5, IH=0
13250987: irq: dio: 0x0 flags: 0x8
13251103: Scheduled job 0x20000968, cb 0x2f19 ASAP
13251263: Running job 0x20000968, cb 0x2f19, deadline 0
13251439: Scheduled job 0x20000968, cb 0x2b7d at 13560480
13560480: Running job 0x20000968, cb 0x2b7d, deadline 13560480
```

On snippets like this attempted join, which clearly states is in the ~900mhz range, the output on TTN console is:

```{
  "time": "2018-05-03T04:46:22.049785276Z",
  "frequency": 867.7,
  "modulation": "LORA",
  "data_rate": "SF10BW125",
  "coding_rate": "4/5",
  "gateways": [
    {
      "gtw_id": "eui-xxxxxxxxxxxxxxxxxx",
      "timestamp": 1132486828,
      "time": "2018-05-03T04:46:21.918571Z",
      "channel": 6,
      "rssi": -127,
      "snr": -8.5
    }
  ],
  "latitude": xxxxxxxxxxxx,
  "longitude": -xxxxxxxxxxxxx,
  "location_source": "registry"
}```

I'm so confused. The metadata is generated by the lora module right, not somewhere in this example code?

matthijskooijman, 2018-05-03:
It looks like there might be a bug in TTN, somehow generating incorrect metadata...

> The metadata is generated by the lora module right, not somewhere in this example code? 

The JSON metadata you get is generated by TTN, based on info it gets from the gateway(s), which is in turn based on the actual frequency and other settings used, which are determined by LMIC. The LoRa module, that is the SX1276 or similar, is fairly dumb and only does as instructed by LMIC.

> The only bright side, is after reinstalling/restarting all applications/gateways, I no longer receive these messages on the gateway.

Not sure what you mean by this. The messages no longer appear in TTN? Or only through other gateways?

gpickney, 2018-05-03:
Thanks for the info. Excuse my ignorance, as this is the first time I'm working with hardware instead of just software.

> The only bright side, is after reinstalling/restarting all applications/gateways, I no longer receive these messages on the gateway.

I wrote this part of the last comment before I posted the picture of the Gateway Traffic. I actually did end up receiving the join requests from the node on my gateway as shown above... albeit on the wrong frequency. You can see that the app/dev euis match up for these requests on the gateway and device traffic logs. However, I only ever saw EV_JOIN_FAILED in my device's Serial logs. The Serial logs on my device also indicated a frequency of ~900 Mhz... not ~860

I'm using the Sandbox Electronics LoRaGo PORT, which is essentially the same hardware as the ic880a. I installed the ic880a software from ttn-zh/ic880a-gateway onto a Raspberry Pi 2 Model B. All went well with the installation and registration on TTN, which indicates its frequency plan is 915mhz.