matthijskooijman/arduino-lmic #173: Heltec ESP32

jpraychev, 2018-05-03:
Hello,

Last few days I am trying to get Heltec ESP32 working with TheThingsNetworks. I am not able to get the OTAA sketch working. When i upload the code to the board i am getting the following message on the serial monitor and after that the board resets and everything starts all over again.

2  A11     : 0x000⸮⸮⸮
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction)
Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1340  PS      : 0x00060530  A0      : 0x800d2e4a  A1      : 0x3ffca560  
A2      : 0x00000003  A3      : 0x3f40107c  A4      : 0x00000002  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d133a  A9      : 0x3ffca540  
A10     : 0x00000002  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Backtrace: 0x400d1340:0x3ffca560 0x400d2e47:0x3ffca580 0x400d0c91:0x3ffca5a0 0x400e6a87:0x3ffca5c0

Stephanv94, 2018-05-03:
I have the same problem using a TTGO ESP32 868Mhz V1.0 board:

`Starting
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction)
. Exception was unhandled.
Register dump:
PC      : 0x400bfdc0  PS      : 0x00060530  A0      : 0x800d305a  A1      : 0x3ffca570  
A2      : 0x00000003  A3      : 0x3f40107c  A4      : 0x00000002  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d0ee2  A9      : 0x3ffca550  
A10     : 0x00000005  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400bfdc0:0x3ffca570 0x400d3057:0x3ffca590 0x400d0ca1:0x3ffca5b0 0x400e6c87:0x3ffca5d0`

jpraychev, 2018-05-05:
@Stephanv94 Hey,

I have compiled another sketch for my Heltec board and I got it working. You can download the example code from https://github.com/jpraychev/Heltec_ESP32

The board I have compiled the sketch for is Heltec ESP32 with SX1276 transceiver. Depending on your board you have to change the radio type in config.h file in the LMIC library.

Another thing to mention is that you have to change the addresses for your particular node. 

Hope this helps you with your project!

If you got any question, feel free to discuss it!

Best regards,
Jordan

Stephanv94, 2018-05-06:
@jpraychev I managed to get the OTAA sketch working setting the pin mapping correctly.

Thanks a lot for your time!

Kind regards,
Stephan

vicatcu, 2018-08-01:
@Stephanv94 how did you actually solve this problem? I'm getting the same exact thing. Does your initialization block look like this?

```c
const lmic_pinmap lmic_pins = {
    .nss = 18,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 14,
    .dio = {26, 33, 32},
};
```

cyberman54, 2018-08-01:
For Heltec32 Board use this lmic pin mapping:

vicatcu, 2018-08-01:
@cyberman54 thanks for the feedback but I still get this bizarre runtime error described in this issue. I wonder what is causing it. Seems like related issues on the web suggest it's the way ISRs are attached. See https://github.com/espressif/arduino-esp32/issues/855.

cyberman54, 2018-08-01:
Lmic is not using interrupts, iz polls instead. And i can assert, the Heltec boards run under arduino and with lmic. The problem will be caused b something in your application, not in lmic and not in the hardware.

lois-lee, 2018-08-01:
@cyberman54 thanks for the input, I was using the example sketch and it was still giving me errors, but adding a delay seems to have fixed the problem. I don't know what went wrong there, but all seems good. Thanks again for the help!

themadsens, 2018-08-10:
I just managed to the the Heltec-LORA-OLED thing going (https://robotzero.one/heltec-wifi-lora-32/) The secret was to merge PR #77  and use the following pin definition:
```
const lmic_pinmap lmic_pins = {
    .mosi = 27,
    .miso = 19,
    .sck = 5,
    .nss = 18,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 14,
    .dio = {26, 33, 32},
};