matthijskooijman/arduino-lmic #176: Debug library missing?

abreups, 2018-05-23:
Documentation mentions a debug library, but I can't seem to find it (nor its functions). Any help?

matthijskooijman, 2018-05-23:
What documentation exactly? There is debug logging support, which can be enabled by modifying `config.h` inside the library.

abreups, 2018-05-23:
File LMiC-v1.5.pdf in section 4.8 talks about a debug library and shows functions like `void debug_init()`, `void debug_led (u1_t val)`, etc. I can't seem to find these functions anywhere.

matthijskooijman, 2018-05-23:
Ah, that PDF is about the original upstream LMIC version, I wouldn't be surprised if it is outdated (I can't recall removing such functions in this Arduino version, but perhaps they were removed by IBM upstream and the documentation stuck around).

abreups, 2018-05-23:
Thanks for the clarification. The document is useful, though. Only this section was "outdated".

closed 2018-05-23 by abreups

abreups, 2018-05-23:
For anyone interested, I found a reference to the original (I think) debug library at http://www.lancaster.ac.uk/scc/sites/lora/lorablinkkit.html
Good luck.