matthijskooijman/arduino-lmic #178: My program stop if I set TX_INTERVAL over 3600

pierrot10, 2018-05-31:
Hello,
I actually have a problem and I have pain to debug or to find the start point to debug it.

I am collecting data sensor and I sent it each hour. It works fine. But know I change the schedule to 2 hours (even 4 hours) and the data are not sent any more.

What I know, the program stop at at the second attend
`LMIC_setTxData2(1, payload, strlen((char *)payload-1), 0);`
In other words, when I lunch the data are sent. Because I setedup
`const unsigned TX_INTERVAL = 7200;  // 2h`
it wait 2 hour until the next send, amd it just before the funtion
`LMIC_setTxData2(1, payload, strlen((char *)payload-1), 0);`
that the program stop

I also tried with
`LMIC_setTxData2(1, payload, sizeof(payload-1), 0);`
But with this, it's not send at the fiirst attend. At least I do not have the message

> 1475691: EV_TXCOMPLETE (includes waiting for RX windows)

The problem I have, I do not know wher to debug because my program has always been working. But it's true, I always schedule the send below 1h.

'payload' is not overloaded.

Do you have any suggestion to start looking into the code , or the library to fix the problem?
Do you have any question which help to support me?

Thank a lot

DeuxVis, 2018-05-31:
Pierrot, not sure this is the root cause but as I read
`strlen((char *)payload-1)`
I understand "length of the string starting 1 byte before payload".

Maybe you want to use instead : 
`strlen((char *)payload) - 1` ?

pierrot10, 2018-05-31:
Dear DeuxVis, Yes, I observed it just after posting my post. I have not replied yet, because I am testing my code and I need to wait two hour. May be it's the cause but in any case, this look wrong. I got a result in about 30mn.

pierrot10, 2018-05-31:
Hello, Unfortunately, it did not solve the problem
```
Si.sprintln(F("hhhhhhh"),2);
        // Prepare upstream data transmission at the next possible time.
        LMIC_setTxData2(1, payload, strlen((char *)payload)-1, 0);
```
My program stopped juste after 'hhhhhhhhh'

pierrot10, 2018-05-31:
Is there a wayto debug the Arduino-LMIC library?
I went to the cofnig file I changed 

`#define LMIC_DEBUG_LEVEL 1`

but I can not compile any more because of this message:

> 
> C:\Users\pierrot\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp:223:54: error: '_FDEV_SETUP_WRITE' was not declared in this scope
> 
>      fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);

As I do have an AVR Feather MO adalogger, I am not sure if I can use this function, but I really want to be able to print message to my terminal, by adding some thing like Serial.println("DEBUG"); in order to tracl the issue

pierrot10, 2018-05-31:
I really wonder if the library can have a TX_INTERVAL  over 3600.
I take that exemple [ttn-abp](https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-abp/ttn-abp.ino) and I only replaced
`static uint8_t mydata[] = "Hello, world!";`
to
`static uint8_t mydata[] = "26,0,364,53,108,0,0,0,0,0,0,0";`

`const unsigned  = 60;`
to
`const unsigned TX_INTERVAL = 7200;`

and adapt
```
// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 5,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 6,
    .dio = {10, 11, 12},
};
```
and after two hours, I have not get a new line as

> 762110: EV_TXCOMPLETE (includes waiting for RX windows)

What is the maximum of TX_INTERVAL ???

gizmocuz, 2018-06-01:
@pierrot10 , could you try to change

const unsigned TX_INTERVAL = 7200;

to

const int32_t TX_INTERVAL = 7200;

gizmocuz, 2018-06-08:
@pierrot10 , did that work ?

pierrot10, 2018-06-08:
Dear gizmocuz,
I just discover your answer 20mn ago. I will try it this evenong éate you know tomorrow as I need at leat two hour to have two measures. But it look interresting. 
By the way, I wonder if I should bette do

`const unsigned long TX_INTERVAL`

At the time writing, I reading this article
[Using millis() for timing. A beginners guide](https://forum.arduino.cc/index.php?topic=503368.0)

> 2.  It is an unsigned long because that is what millis() returns.
> 3.  It is important that it is unsigned (more on this later)

I come back to you tomorrow, but I will try the two solution (int32_t, uint32_t and lon)

gizmocuz, 2018-06-08:
@pierrot10 , actually the best is to define it, like

`#define TX_INTERVAL 7200`

I see it is used in this function, which returns a int64_t function:

#define sec2osticks(sec) ((ostime_t)( (int64_t)(sec) * OSTICKS_PER_SEC))

Hope it works ! Thats some long delay between transmits ;)

pierrot10, 2018-06-09:
Appently none ofthis work.
I tried
`const unsigned long TX_INTERVAL`
and
`#define TX_INTERVAL 7200`
The program bug or stop running just before the enxt send
`LMIC_setTxData2(1, payload, strlen((char *)payload)-1, 0);`

I can not understanf how work the intervals TX_INTERVAL

> Thats some long delay between transmits ;)
I ma measuring the soil moisture and I do not need to send the easure each minute. Each two hours is enough and I am planning to measure ech 4 hours.

Apparently it's work if I send each hour, but not each I extend the intervals.

Any idea what happen?
Should I try with
`const int64_t TX_INTERVAL`
but that will really make the difference?

Thank for helping