matthijskooijman/arduino-lmic #180: FAILURE error using Arduino Uno & SX1276

lois-lee, 2018-06-12:
Hi! 

So to start off, I'm using the ttn-otaa sketch with some minor modifications. 
When I run the code however, it starts, the packet is queued, EV_JOINING...
and then I get some unknown events after which it successfully joins, but then I get
the FAILURE error like so:

[image]

I've tested my pin connections and I know that I put in the keys correctly since it has connected and the device does show on the ttn console. The line that it crashes on (lmic line 1541) is the following:
 
ASSERT((LMIC.opmode & OP_TXRXPEND)!=0);

But I'm not sure if this should really be a problem since I haven't changed much with the configuration to begin with, and theoretically the sketch should run with minor changes to the config file and the sketch itself. Here are the pin connections for reference:

const lmic_pinmap lmic_pins = {
    .nss = 6,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {3, 4, 2}, 
};

Some other weird symptoms it's been showing are:
-> Sometimes it fails to connect but still shows up on the console, only successfully "JOINED" network when I'm in an area covered by 4 gateways (and I have an antenna)

Some things I've went over and already tried:
-> Looking over the config file and uncommenting the PING disable, checking the radio model was correct, checking the country code and radio freq are correct.
-> Tested pin connections and antenna connection.

matthijskooijman, 2018-06-12:
I believe I've seen similar behaviour on a bad connection. Perhaps you have a bad solder joint somewhere that works sometimes but not always? Does it fail at the same point every time, or is it random?

lois-lee, 2018-06-12:
It seems to be consistently at the same line. My connections should be okay, I tested them with a multimeter. I can recheck them, but are there any other reasons this could be happening?

cyberman54, 2018-06-12:
All similar cases usually turned out to have issues with cabling of DIO0/1 lines, e.g. broken or wrong wiring, or the GPIO port is connected to some other components, too.

lois-lee, 2018-06-12:
Hmm, well I checked the DIO0/1 lines with a multimeter from the soldered pin to the arduino input, so it should theoretically be correct, but when I just switched out the jumper cables with some single core wire, it won't even join (aka EV_JOIN_FAILED). 

I feel like it could definitely still be a problem with the connections like both of you have suggested, so I'm going to over them another time, but also, I mean I'm getting a really weak rssi even when I'm in between two gateways, and even now when I'm in the scope of 4 gateways, I'll still fail to join.

I'm not using any other components, so I feel like it also couldn't be that either...I only have the SX1276 hooked up to the arduino.

lois-lee, 2018-06-12:
Here's what it's outputting most of the time:
[image]

cyberman54, 2018-06-12:
You need to check what causes those unknown events. You shouldn't have any of them.

lois-lee, 2018-06-12:
Okay, yeah, I also thought that the unknown events were problematic, but I guess I should look more into it. I'm not exactly sure why they're occurring. I know that it's just defaulting because the event doesn't match any of the previous cases, but I still need to look into that more.

cyberman54, 2018-06-12:
I think all these issues are caused by undefined or unstable signals on your DIO0, DIO1, DIO2 pins.
Don't use jumper cables on breadboards.

closed 2018-06-13 by lois-lee