matthijskooijman/arduino-lmic #182: Acknowledgment

carabaf, 2018-06-13:


carabaf, 2018-06-13:
I have two nodes connected to my gateway (dual_chan_pkt_fwd_up_and_down) and I send confirmed message. After several transmission succesfully acknowledged, one node can't anymore receive ack messages. Then, it takes several minutes before a new transmission is sent.

DeuxVis, 2018-06-13:
I'm not 100% sure about this, but I believe that for 2 ways communications a single or dual channel gateway is not sufficient : the network is expecting a fully capable gateway and might ask yours to answer your nodes on a channel it cannot actually use.

Are you using the things network ? You should check on their forums, some much knowledgeable people will be able to give you a definitive answer on that.

carabaf, 2018-06-15:
I will check about that. I am using loraserver so I will check on their side.

closed 2018-06-15 by carabaf