matthijskooijman/arduino-lmic #190: Adafruit Feather 32U4 LoRa device USB bricked on 2nd upload

rlramirez77, 2018-06-30:
1st code upload had no problems and the device was sending packets through my gateway, but when I made a minor change and uploaded again, Windows (10) reported that it could not recognize the device. There's a unknown USB device with the error (Device Descriptor Request Failed). I tried Zadig and no go, then I tried a knock-off of the same 32U4 LoRa board and the same thing happened.

NavaAntonio, 2018-08-13:
Hi, I also have a Feather 32u4 and that problem that you mention happens very often,  i´m not sure why happens, but I have solved simply adding while (! Serial) after the Serial.begin (baudRate).

I have also experienced that windows does not recognize my feather32u4 if it has this problem, these are the steps that have worked for Windows to recognize it again.

1. First for windows to recognize the new usb port it is necessary to connect the feather32u4 and go to Tools-> Port and windows will assign COM to the Feather32u4. If no COM appears, then press reset until a COM is assigned (in my case it is usually COM4).

2. After the COM is assigned it is necessary to load a program such as flashing. click on load and when in the process of loading, press reset two times. The program should load on the feather32u4. (steps described in "Bootloading manual" at https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/using-with-arduino-ide)

3.Windows should detect the Feather32u4 (as COMX (Adafruit Feather 32u4)). After that just add the while (! Serial) after the Serial.Begin and it should not present problems.

Garfius, 2019-07-23:
> Hi, I also have a Feather 32u4 and that problem that you mention happens very often, i´m not sure why happens, but I have solved simply adding while (! Serial) after the Serial.begin (baudRate).
> 
> I have also experienced that windows does not recognize my feather32u4 if it has this problem, these are the steps that have worked for Windows to recognize it again.
> 
> 1. First for windows to recognize the new usb port it is necessary to connect the feather32u4 and go to Tools-> Port and windows will assign COM to the Feather32u4. If no COM appears, then press reset until a COM is assigned (in my case it is usually COM4).
> 2. After the COM is assigned it is necessary to load a program such as Blink example. click on load and when in the process of loading, press reset two times. The program should load on the feather32u4. (steps described in "Manually Bootloading" https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/using-with-arduino-ide)
> 
> 3.Windows should detect the Feather32u4 (as COMX (Adafruit Feather 32u4)). After that just add the while (! Serial) after the Serial.Begin and it should not present problems.

Already uploaded blink, and worked.

Now the usb port looks dead --> USB\DEVICE_DESCRIPTOR_FAILURE

Totally bricked

pmethods, 2019-07-30:
while (! Serial); 
That solved the issue for me.  Just added the line after Serial.begin(115200);

Issue = "Uploads sometimes, but irritating and inconsistent."

-methods