matthijskooijman/arduino-lmic #192: OTAA mode - SX1276 - EU868

sephiroth6, 2018-07-31:
Hello,
I have a problem with EV_JOINING, there is an infinite loop:
[https://pastebin.com/TBhTAFXk](url)
I already tried to find an issue (closed and open) with my same problem, but everyone have something with pins or deveui setting wrong.
I use arduino uno with dragino shield 1.4, config.h setted with EU conf.

Pins map:
> const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 6, 7},
};

I use the otaa raw for initial config. I tried to switch the byte of deveui and appeui as someone said for least-significant-byte first, but same result in Serial.

Some override param:
`LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);`
`LMIC.freq = 868100000;`

What can I do?
Thx!

cyberman54, 2018-08-01:
For me this looks like your node just does not join the network, for some reason. Can't see any strange behaviour in the log. Did you test it in a second gateway setup?

sephiroth6, 2018-08-01:
Yes, same problems. I just can connect with ABP mode

cyberman54, 2018-08-01:
Then you should check the timings. Maybe your node is too slow.