matthijskooijman/arduino-lmic #198: EV_TXCOMPLETE regardless of duty cycle limit

JackGruber, 2018-09-02:
Hi,

I always get EV_TXCOMPLETE back when I call do_send(&sendjob). 
Even if the message was not sent, because there was no window to send due to duty cycle limitations?!

Tested with the ttn-abp.ino, TX_INTERVAL = 60 and only one channel.

Is this normal?

matthijskooijman, 2018-09-02:
Nope, it should wait for airtime to become available, then send, then receive and only then trigger TXCOMPLETE. Are you using the unmodified library from this repo?

JackGruber, 2018-09-02:
Yes, the problem occurs with a ESP32 TTGO and a Arduino ProMini and a RFM95 module.

JackGruber, 2018-09-02:
Attached the log output, only the first and last message arrived at the gateway.

[console_debug.log](https://github.com/matthijskooijman/arduino-lmic/files/2343464/console_debug.log)

JackGruber, 2018-09-03:
Okay, I think I found the problem. 
I only commented out the channel definition, but didn't disable the channels with LMIC_disableChannel.

closed 2018-09-03 by JackGruber

BThunderman, 2019-12-12:
JackGruber, can you please clarify what you mean with disabling the channels? I have the exact problem and don't know the problem.

JackGruber, 2019-12-12:
@BThunderman

My problem was, I just commented out the channel definition and not disabled the channels.  
Call `LMIC_disableChannel(CHANNEL_NR); ` for ever channel you want do disable. 

Meanwhile I am using another LMIC implementation which receives updates. 
https://github.com/mcci-catena/arduino-lmic

BThunderman, 2019-12-16:
Thank you very much, now it works!
Yes I use the MCCI one too, but had same error there.