matthijskooijman/arduino-lmic #199: FAILURE radio.c:818

jmppCT, 2018-09-05:
Hi,  I`ve been working in a  node using RFM95W connected to Arduino Nano Board using a a 9V battery with a voltage regulator to supply the 3.3V to the RFM95
I've already tested connection between the node RFM95 and the gateway and it was working fine some time.
Now, I have change the 9V battery for 2 x 3.7V  18250 Li-ion Rechargeable Batteries with the same voltage regultator that supply the 3,3V .
I´m testing the node and the  systems return the next failure:
C:\Users\JOSE\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:818

Does anybody has the same problem ? 
I would be greatful for any information to solve it.

Thanks in advance
Jose

utkarshshah007, 2018-09-06:
That method is eventually called by `hal_enableIRQs();` in oslmic.c, to create interrupts.
Not sure what your issue is, but I'd add a couple print statements to see if you can follow it back to the root cause.

jmppCT, 2018-09-07:
Thank you very much, 
you're right. The problem is related to IRQ.
Following the code line 818 is inside radio_irq_handler (u1_t gave)  concretely    ASSERT(0);
I have tested again  the module with the power from the 3.3 V pin of arduino and does not indicate any error (EV_TXCOMPLETE (includes waiting for RX windows)  but the gateway does not receive any data.
I´m going to review all the wiring and the code again .

NathanGanesh, 2021-03-16:
how did you fix this?