matthijskooijman/arduino-lmic #201: Transmission is sending, but I'm not reciving on the gateway/server end

6d6176, 2018-09-18:
915 freq, 

Packet queued
96386060: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
96829613: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
97273165: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
97716714: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
98160266: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
98603888: EV_TXCOMPLETE (includes waiting for RX windows)

miguerman, 2018-11-12:
Hi.
I have the same problem, I have TTGO ESP32 LORA V2, I have configured the mapping pins, and the hal.cpp for SPi.
I am using the ABP example.
Can somebody help me.
Greetings.

trlafleur, 2018-11-12:
Here is my code that works for OTAA mode, has been running for a few months... you need to change the APPEUI and DEVEUI to match your application on TTN.

It runs on a TTGO ESP32 LORA V2 with OLED US915MHz
IDE 1.8.5

(It has a number of #ifdef for other processors and my special needs, stuff you may want to strip out... but I did not have the time to clean it up.... It run's....)

~~~

/*
 * TRL  ver 1.1f   29 Jun 2018
 *  added: ESP32 + OLED Display 3 Mar 2018
 * 
 * 
 *  Test Code running on these processors:
 *    TTGO ESP32
 *    Rocketscream ARM M0 RFM95 LoRa board
 *    MoteinoMega LoRa
 * 
 * 
 *  Notes: TTN Port numbers of 1-223 are legal for application
 * 
 * 
 * 
 * 
 * 
 * */

 /* *******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello, world!", that
 * will be processed by The Things Network server.
 * 
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in g1, 
 *  0.1% in g2). <--- Not in US
 *
 * Change DEVADDR to a unique address! 
 * See http://thethingsnetwork.org/wiki/AddressSpace
 *
 * Do not forget to define the radio type correctly in config.h, default is:
 *   #define CFG_sx1272_radio 1
 * for SX1272 and RFM92, but change to:
 *   #define CFG_sx1276_radio 1
 * for SX1276 and RFM95.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>
#include <stdarg.h>
//#include <gBase64.h>
#include <Streaming.h>
#include <Wire.h>         // http://arduino.cc/en/Reference/Wire ??

/* ************************************************************************************** */
// User defines
//#define OLED              // If we are using OLED display
[... 652 lines cut ...]
    debug(PSTR("\n***  LMIC_StartJoining \n"));
    JoinedFlag = false;
    LMIC_startJoining();

// Start job's (sending automatically starts OTAA too), they will re schedule themselves
    do_send_task1(&sendjob);
    
    do_keepalive(&keepalivejob);
    //os_setTimedCallback(&keepalivejob, os_getTime()+sec2osticks(KeepAlive_TX_INTERVAL), do_keepalive);
}

// end of setup
/* ************************************************************************************** */

/* ************************************************************************************** */
/* ************************************************************************************** */  
/* ************************************************************************************** */  
void loop() 
{
    os_runloop_once();      // run the LMIC scheduler

 }
/* ************************************************************************************** */

~~~

SandroMoretti, 2020-06-17:
@6d6176 @miguerman I'm having saming issue, using esp32 ttgo ABP mode.
Did you got this?

tortuderelii, 2021-09-21:
@SandroMoretti Hey, did you solve the problem? I have the same issue.

CastilloDelacroix, 2021-11-08:
Nobody can solve this? I have the same problem

trlafleur, 2021-11-08:
I highly suggest you move on to MCCI version as the this basic LMiC code is no longer supported and is way out of date.

You may also want to look at Seeed E5 module that has LoRa stacks, CPU and radio in one module. (Makes using LoRa easy)

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~