matthijskooijman/arduino-lmic #202: Arduino UNO/MEGA + Dragino LoRa Shield v1.4 shield: Class B Support (LoRaWAN 1.0.3)

mhornjak, 2018-09-21:
Hi everyone.
Did anyone successfully try to implement LMIC on an Arduino UNO or MEGA with Dragino LoRa Shield v.1.4 (SX1276) Class B?
I have loraserver.io working and emitting Beacons on 869525000 Hz, but Beacons do not get recognized.
However, Class A Join (ABP) works fine, uploading of data in class A works fine.
What about Class B? In the Readme it says Class B is fairly complete implemented but not tested. 
Anyone got experience on this?

Or even an alternative?
Thank you very much, support is really appreciated!

matthijskooijman, 2018-09-24:
I have personally never tried Class B, I've only been running with TTN which does not support it. LMIC *should* support Class B, and I am not aware of any changes that break support, but I never tried it.

kierenblack, 2018-09-28:
I tried to get Class B to work but it just stopped transmitting with opcode 0xc09