matthijskooijman/arduino-lmic #211: Received payload is different from "send" payload

itofficeeu, 2018-11-29:
Thanks for a awesome library!

During testing sending data to the backend, I always send the same small set of data.

`    
    buffer[0] = (char) 9;
    buffer[1] = (char) 29;

    Serial.print("celciusInt"); Serial.println(celciusInt);
    Serial.print("buffer[0]="); Serial.println(buffer[0]);
    Serial.print("buffer[1]="); Serial.println(buffer[1]);
    Serial.print("sizeof(buffer)="); Serial.println(sizeof(buffer));

    // Prepare upstream data transmission at the next possible time.
    //LMIC_setTxData2(1, lpp.getBuffer(), lpp.getSize(), 0);

    // Send on port 1, without asking for confirmation:
    LMIC_setTxData2(1, buffer, sizeof(buffer), 0); // 0x0929 for 23.45
`

At TTN I get totally different data in the payload. The data aren't even the same each time I send data. They change each time they are send. The 'funny' thing is, they make a series of data for each restart of the microcontroller.

Take a look at this:

Restart Number         1       2       3    ...

Payload in TTN

           count
               0           BF90    BF90    BF90    ...
               1           3581    3581    3581    ...
               2           4DC7    4DC7    4DC7    ...
               3           7CC9    7CC9    7CC9    ...
              ...          ...

Are the data mixed with the count?...

Am I doing something wrong?

Thanks :)

matthijskooijman, 2018-11-29:
Now that's weird... Are you using OTAA or ABP? And you're just using the regular TTN backend?

My first guess would be invalid encryption keys, which could lead to incorrect decryption. However, if you would configure the wrong keys, then the checksum and other header data would also be messed up and data would never reach your application in TTN.

Can you run with debug level 2 (see config.h in the library) and paste the results here? Perhaps there's a hint in there...

itofficeeu, 2018-11-29:
I have further information.

I reset the program to use Cayenne.lpp AND use up to 51 bytes. - Like this:

`
#include <CayenneLPP.h>
CayenneLPP lpp(51);
`

`
lpp.reset();
lpp.addTemperature(1, 23.45);

LMIC_setTxData2(1, lpp.getBuffer(), lpp.getSize(), 0); // 0x0929 for 23.45

`

It now spits out sort of random data like before, even if I always just tries to send the same value '23.45'. But by accident does it produce usable data. Not the data I send, but still some sort of 'usable' data.

The 'usable' data are these

Payload: 2C 65 EA 98

Fields:
{
  "luminosity_44": 60056
}

This should actually be 23.45 hat I was sending. But if I look the this link:
https://mydevices.com/cayenne/docs/lora/#lora-cayenne-low-power-payload

I can convert the EA 98 to 60056,

And the table over Data Types tells me that '65' in front of the value stands for Illuminance Sensor - In my case named 'luminosity_44* by something I don't control.

I do not yet know what 2C stands for. - maybe the channel?...

Another 'usable' data was:

Payload: 32 03 92 98

Which again according the Cayenne data types can be transferred to:
{
  "analog_out_50": -280.08
}

itofficeeu, 2018-11-29:
More information;

I have now tried to make a buffer like a Cayenne format - like this:

`
    uint8_t buffer[4]; // reserve 4 bytes in memory

    buffer[0] = 0x00; // - I don't know hat to put.
    buffer[1] = 0x67; // For the data type Temperature Sensor
    buffer[2] = 0x09; // First part of the float (MSB)
    buffer[3] = 0x29; // Second part of the float (LSB)

    Serial.print("buffer[0]="); Serial.println(buffer[2]);
    Serial.print("buffer[1]="); Serial.println(buffer[3]);
    Serial.print("sizeof(buffer)="); Serial.println(sizeof(buffer));

    // Send on port 1, without asking for confirmation:
    LMIC_setTxData2(1, buffer, sizeof(buffer), 0); // 0x0929 for 23.45
`

But this does not lead to any usable or 'usable' data. Just randon like the first time.

matthijskooijman, 2018-11-29:
I had a look over your code and output, but could not find anything obviously wrong. I'm a bit at a loss here, unfortunately.

Did you try the ABP example sketch as-is? Does that work?

matthijskooijman, 2018-11-29:
Good to hear there's some progress there :-)

If you find the problem, care to report here? I'm curious what could cause this erratic behaviour...

itofficeeu, 2018-11-29:
Hunted down to the APPSKEY having just one single letter/number wrong.

Unfortunately is the font used in Arduino IDE not clear in the difference between 8 and B. :(

- This can only be lead back to somebody loving a "good" looking font more, than something that is readable. :(

This has "only" cost me 3 days. At least it works. Time to move away from Arduino IDE. What a crap.

itofficeeu, 2018-11-29:
@gizmocuz - Wow! Thanks! Nice! :D Wonderful link! :D  & thanks to @matthijskooijman too for being here! :D