matthijskooijman/arduino-lmic #213: HelTec WiFi LoRa 32 - v2/Version 2 - Not sending dispite of same software as v1

itofficeeu, 2018-12-01:
The version 2 of the hardware does not send data as version 1 of the hardware, even if the software is 100% the same.

The program runs and sends one LoRa message (once) without any payload. The existence of the single message can be seen in the data section of the application in thethingsnetwork.org.

The hardware continues to run but does not send any further LoRa packages and data. The continued running can be seen by the software printing debug messages as normal in the monitor of the Arduino IDE.

Before stopping sending data the following debug can be seen in the monitor:

`Starting...
RXMODE_RSSI
LMIC setup done!
LMIC.opmode= 2048
OP_TXRXPEND= 128
14683: engineUpdate, opmode=0x808
14700: Uplink data pending
14703: Considering band 0, which is available at 14673
14924: Considering band 3, which is available at 0
15207: No channel found in band 3
15396: Considering band 0, which is available at 14673
15700: No channel found in band 0
15890: Considering band 1, which is available at 14673
16194: Airtime available at 14673 (channel duty limit)
16498: Ready for uplink
16650: Updating info for TX at 14699, airtime will be 2896. Setting available time for band 1 to 304299
17225: TXMODE, freq=868100000, len=13, SF=7, BW=125, CR=4/5, IH=0

Time    ; Date      ;Satelites/# ;Fix;Quality;Location/dg        ;Speed/knots;Speed/km/h; Angle  ;Altitude/m
48788: irq: dio: 0x0 flags: 0x8
48865: Scheduled job 0x3ffc3b98, cb 0x400d31d4 ASAP
49150: Running job 0x3ffc3b98, cb 0x400d31d4, deadline 0
49465: Scheduled job 0x3ffc3b98, cb 0x400d2edc at 111257
111257: Running job 0x3ffc3b98, cb 0x400d2edc, deadline 111257
111384: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
12:03:40; 2018.01.12;           4;  1;      1;  37.5006,  -1.0749;       1.30;      2.41;   13.04;      23.40
12:04:00; 2018.01.12;           4;  1;      1;  37.5005,  -1.0747;       1.65;      3.06;  348.22;      23.40
12:04:20; 2018.01.12;           4;  1;      1;  37.5003,  -1.0748;       1.15;      2.13;  149.31;      23.40
`

After this, the lmic.c software stops printing any more debug massages. Only messages from your own code will be printed in the monitor.

I will now take a look at the settings for the LoRa module in version 2 of the hardware.

itofficeeu, 2018-12-01:
I just tried to change to the repo "IBM_LMIC_framework". 

Not forgetting to disable

`//#define LMIC_PRINTF_TO Serial`

in config.h.

Here does the error occur in the same way. You just do not have the same debug output available. So I can not give you any based in that repo.

itofficeeu, 2018-12-01:
Heltect has changed the pinout for the LoRa sender module.

The settings for LMIC for version 2 of the hardware is now:

`// Pin mapping according to Cytron LoRa Shield RFM
const lmic_pinmap lmic_pins = {
  .nss = 18,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 14,
  .dio = {26, 34, 35},
};`

Version 1 has "32, 33" - where version 2 has "34, 35" as pins for the Lora module.

v1 pinout here:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32(V1).pdf

v2 pinout here:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32(V2).pdf

closed 2018-12-01 by itofficeeu

Heltec-Aaron-Lee, 2018-12-03:
This page have a detail document about the difference between V1 and V2:
[](http://www.heltec.cn/wifi-lora-32-new-edition-comparison/?lang=en)
[image]

BTW. you can try our LoRaWAN library: [](https://github.com/HelTecAutomation/ESP32_LoRaWAN)

Heltec-Aaron-Lee, 2018-12-03:
This page have a detail document about the difference between V1 and V2: [http://www.heltec.cn/wifi-lora-32-new-edition-comparison/?lang=en](url)
[image]

BTW. you can try our LoRaWAN library: [https://github.com/HelTecAutomation/ESP32_LoRaWAN](url)

Humancell, 2018-12-03:
@Heltec-Aaron-Lee your website is not working ... and I am only getting 504 Gateway Errors from the USA. Is there a problem with your webserver?

Heltec-Aaron-Lee, 2018-12-04:
@Humancell the web server is working well