matthijskooijman/arduino-lmic #215: Setting spreading factor

DefProc, 2018-12-18:
What is the intended way of setting a fixed spreading factor?

Should this be done in the library files, or in the Arduino Sketch?

jmessiasalves, 2021-02-28:
Hi,

I am using AU915. How to set DR_SF8 with BW 500kHz?
When set with LMIC_setDrTxpow (DR_SF8, 14) the BW is only 125kHz.

Thank you in advance.

Att.,

matthijskooijman, 2021-03-01:
IIRC this version of LMIC does not support AU915 at all, maybe you are using another version and are asking in the wrong place?

Also, does the AU915 spec actually define a SF8+500kHz datarate? If not, it might not be easy to use it, since then there is no DR identifier for it and you might need to manually set the rps value (and maybe change LMIC to not overwrite it) or change LMIC to add a custom DR somehow.

jmessiasalves, 2021-03-02:
@matthijskooijman ...thanks for listening. Sorry for the post in the wrong place. I'm using the MCCI_LoRaWAN_LMIC_library which offers the AU915 frequency plan.
Anyway, thank you very much indeed.