matthijskooijman/arduino-lmic #216: Function os_getBattLevel

kamarat, 2019-01-08:
Hello,
I've used arduino-lmic library for my sensor node. I would like to implement function _os_getBattLevel_ correct without edit the library files.

There is pieces of code in file _/src/lmic/oslmic.h_
`#ifndef os_getBattLevel`
`u1_t os_getBattLevel (void);`
`#endif`
and in file _/src/lmic/lmic.h_
`#if !defined(os_getBattLevel)`
`u1_t os_getBattLevel (void) {`
`    return MCMD_DEVS_BATT_NOINFO;`
`}`
`#endif`

When I wrote own function _os_getBattLevel_ and commented code in file _arduino-lmic/src/lmic/lmic.h_, that was working.

But when I wrote macro  
'#define os_getBattLevel() getBattLevel() '
and new own function _getBattLevel()_, that wasn‘t working.
What is correct implementation?

sle118, 2019-04-03:
This question has more to do with understanding of compiler and C programming than it has to do with this library. I suggest you create a repository, upload you code there, and describe how you compile the code (Arduino compiler vs PlatformIO, etc) as there are many ways you could get this problem and each environment had is own quirks.

Perhaps you should consider getting help from one of the many forums online; there are many active communities out there that are likely to help. Issues in GitHub should be strictly related to the project itself. For example feature requests, actual bugs that are fully documented, etc.

cyberman54, 2019-09-07:
I ran in exactly the same problem. Tried several option (forward declaration, etc.) but did not manage to declare own function os_getBattLevel without modify the source code of the library. Is there any straight forward solution for this?

geeks-r-us, 2020-09-07:
Issue is within the "hacked" symbol resolving process to allow lazy code writing without creating prototypes etc. 
In PlatformIO you can resolve it by adding the define to platfomio.ini;
```
build_flags = 
  -D os_getBattLevel=getBattLevel
```
I don't know if there is any chance to do something like that with the Arduino IDE