matthijskooijman/arduino-lmic #218: radio.c:750 - no idea what's wrong

E2w4UmWh, 2019-02-05:
Hello all,

I'm running a dragino LoRa Shield v1.4 on an Arduino Uno on 433 MHz.
Arduino reports me the following error:
FAILURE  ...\radio.c:750

I have no clue, where to start from the debugging, as I do not understand the goal of the code around line 750. Any hint, how to proceed is welcome ...

dl1mtk, 2019-02-09:
Arduino Reports me the same error.
I use the 868 MHz Version.

:-(

matthijskooijman, 2019-02-10:
What version are you using? The (latest) version in this repo does not have any assert in that particular line: https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L750

Perhaps you could look up the corresponding line (and a few lines around it) in your version and paste it here? That should give some hint about what's wrong.

dl1mtk, 2019-02-11:
That is suspect. 
I think I have the lastest version. Downloaded 2 or 3 days ago from github.

But in my line 750 is something different
I marked it (right side).

I will check it again later

Thx Holger

`  // Sets a Frequency in HF band
    u4_t frf = 868000000;
    writeReg(RegFrfMsb, (u1_t)(frf>>16));
    writeReg(RegFrfMid, (u1_t)(frf>> 8));
    writeReg(RegFrfLsb, (u1_t)(frf>> 0));

    // Launch Rx chain calibration for HF band
    writeReg(FSKRegImageCal, (readReg(FSKRegImageCal) & RF_IMAGECAL_IMAGECAL_MASK)|RF_IMAGECAL_IMAGECAL_START);
    while((readReg(FSKRegImageCal) & RF_IMAGECAL_IMAGECAL_RUNNING) == RF_IMAGECAL_IMAGECAL_RUNNING) { ; }
#endif /* CFG_sx1276mb1_board */

    opmode(OPMODE_SLEEP);

    hal_enableIRQs();
}

// return next random byte derived from seed buffer
// (buf[0] holds index of next byte to be returned)
u1_t radio_rand1 () {
    u1_t i = randbuf[0];
    ASSERT( i != 0 );                                                                                 // That is Line 750
    if( i==16 ) {
        os_aes(AES_ENC, randbuf, 16); // encrypt seed with any key
        i = 0;
    }
    u1_t v = randbuf[i++];
    randbuf[0] = i;
    return v;
}`

matthijskooijman, 2019-02-12:
Hm, haven't seen that error before. It's asserting about an internal randomization variable, that really should never be zero. Two possible causes spring to mind:
 - Did you perhaps forget to call `os_init()` on startup?
 - Are you perhaps running out of RAM? If that happens, the stack variables might end up overlapping (and thus overwriting) the global variabels, which could lead to random behaviour like this.

moferreira, 2019-07-02:
I am new to IoT/Lora so I got a Dragino Lora IOT Kit V2. I am getting the same message at the serial monitor:

Connect to TTN and Send data to mydevice cayenne(Use DHT11 Sensor):
###########    COUNT=1    ###########
The temperature and humidity:
[0.00℃,0.00%]
FAILURE 
/root/Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:750

Any help? Please consider I am a beginner.
Thanks.