matthijskooijman/arduino-lmic #223: Error: arduino-lmic-master/src/lmic/radio.c:660

mdrenth76, 2019-04-20:
Hi,

Today I managed to get an arduino Uno working with a rfm95 module to the ttn network.
Two packets where succesfully receivced.

After that everything stopped working and I got these messages:

21:52:27.853 -> Starting
21:52:27.853 -> Packet queued
21:52:29.890 -> 126906: EV_TXCOMPLETE (includes waiting for RX windows)
21:53:29.901 -> Packet queued
21:53:29.937 -> FAILURE 
21:53:29.937 -> /Users/michel/Documents/Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:660

Somebody any ideas what could go wrong?

Best regards,

Michel.

matthijskooijman, 2019-04-23:
I think you're triggering this assertion: https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L660

That means that when the library wants to start listening, it finds that the chip is not sleeping as it should be. I recall this assertion being triggered before, but I can't quite remember the cause. Did you look for existing similar issues already?

mdrenth76, 2019-04-27:
Hi Matthijs,

Thanks for your reply. I rewired everything from start and now it works like a charm. I'm seeing my data in the TTN console. In the process I got some errors on multiple lines of the same radio.c.
Seems like all those errors are due to connection issues.

Thanks for your support!

mdrenth76, 2019-04-27:
btw... great job Matthijs, creating this lib.

kontaktbl, 2019-09-14:
Hi Matthijs,

I have same (or similar) problem - failure lmic\radio.c:660

I have one rfm95 module and it works fine.
But I ordered two new modules (from same aliexpress seller) and both modules reported Failure in line 660 during (or after) JOINING: module sent 'JoinRequest' and asserted on 'JoinAccept'  (I can see it on network server).
log out:
5 Sending.. Packet queued
35 Waiting.. EV_JOINING
...FAILURE
C:\Users\User\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:660

Just to be clear, I have 3 nano devices and 3 rfm95 modules: only one rfm95 works with any of 3 nanos and 2 rfm95's don't work with any nano.
I was try to reconnect wires but not to solder again.

So, what do you think?
thanx,
Sasa

kontaktbl, 2019-11-15:
> Hi Matthijs,
>
> I have same (or similar) problem - failure lmic\radio.c:660
>
> I have one rfm95 module and it works fine.
> But I ordered two new modules (from same aliexpress seller) and both modules [...]

wires from rfm95 to Nano were too long for 8MHz (divided by 4)!!!
Thnx
S.

kahanchudasama, 2020-02-29:
> Hi Matthijs,
> 
> Thanks for your reply. I rewired everything from start and now it works like a charm. I'm seeing my data in the TTN console. In the process I got some errors on multiple lines of the same radio.c.
> Seems like all those errors are due to connection issues.
> 
> Thanks for your support!

HI,
currently i am facing same issue can you guide me how can you solve this?

xNestinhox, 2021-03-19:
C:\Users\nesto\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:535

Buenas colegas tengo este error, estoy usando un sensor hx711 + lora shield con TTN, el codigo en la compilacion no muesra error pero al ver en monitor serie presenta ese mensaje 

a que se refiere este error