matthijskooijman/arduino-lmic #225: LMIC with Low Power

CongducPham, 2019-05-10:
Hi, I tried to make LMIC working with low power mode when timer0 is not updated. After several attempts I ended up to modify hal_ticks() in order to take into account a sleep value that is updated from the main device sketch with a new function (https://github.com/CongducPham/LowCostLoRaGw/blob/528f97b3a4fcc61912d06e6ed3f37073208eab52/Arduino/libraries/lmic/src/hal/hal.cpp#L222). The code is here:

https://github.com/CongducPham/LowCostLoRaGw/blob/master/Arduino/libraries/lmic/src/hal/hal.cpp

It seems to work fine so far, but I'm still worried about possible rollover and I would appreciate your comment, especially on line https://github.com/CongducPham/LowCostLoRaGw/blob/528f97b3a4fcc61912d06e6ed3f37073208eab52/Arduino/libraries/lmic/src/hal/hal.cpp#L149

best regards,

Emielcp, 2019-07-26:
Hii,  

do you have an example of your main sketch using this method? Thanks!

CongducPham, 2019-07-26:
You can find these info here, as well as the last version of the LMIC modifications.

https://github.com/CongducPham/LMIC_low_power

best regards,

matthijskooijman, 2019-07-26:
In one project I updated the `millis()` counter rather than the LMIC ticks, which is more general (but also needs to access internal Arduino variables). See https://github.com/meetjestad/mjs_firmware/blob/master/mjs_firmware.ino#L232-L242

Emielcp, 2019-07-26:
Thank you Pham,

I need a solution for the teensy 3.2 (ARM Chip), I will try this

CongducPham, 2019-07-26:
@matthijskooijman Thanks, is the hack on millis() working on all platforms?

Emielcp, 2019-07-26:
@CongducPham I find the example file (temp) a bit confusing. I'm using a standard LMIC code (like the examples) with the snooze library for my teensy. You use them in the example file as well. Could you have me with my code? Would be awesome! I just need the timer to work again after going in hibernate mode. For code see:

Teensy forum: https://forum.pjrc.com/threads/56999-Teensy-3-2-Snooze-LoRaWAN-LMIC-stack-problem

or TTN forum: https://www.thethingsnetwork.org/forum/t/teensy-3-2-snooze-lorawan-lmic-stack-problem/28411

My mail is: emielcpnl@gmail.com

Thanks in advance!

CongducPham, 2019-07-26:
@Emielcp as indicated in the README, you need the modified LMIC lib. Then, in your code, after wake-up, just insert "hal_sleep_lowpower(8);" where "8" here is the hibernate duration that you were using. Can be something else depending on your code. 

On the Teensy, I actually don't remember if the deep sleep mode preserves the internal clock or not. If millis() counter is preserved then you may not need the LMIC hack.

Emielcp, 2019-07-26:
Okay i tried, still same result. It goes into hibernate and return every 20sec but my message arrives only every 3min and 20sec. Could you please check? Code: 

`#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>
#include <Snooze.h>
#include <i2c_t3.h>

// LoRaWAN NwkSKey, network session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
static const PROGMEM u1_t NWKSKEY[16] = { 0x09, 0x47, 0x9E, 0x7C, 0xA2, 0xEF, 0x71, 0xFB, 0x11, 0x03, 0x85, 0x53, 0x52, 0xF0, 0xEB, 0x0A };

// LoRaWAN AppSKey, application session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
static const u1_t PROGMEM APPSKEY[16] = { 0xC0, 0x5B, 0x13, 0x22, 0x55, 0xC2, 0xC4, 0xF0, 0x30, 0x34, 0xE7, 0x72, 0x6A, 0x81, 0xF4, 0xC5 };

// LoRaWAN end-device address (DevAddr)
static const u4_t DEVADDR = 0x2601190B ; // <-- Change this address for every node!

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui (u1_t* buf) { }
void os_getDevEui (u1_t* buf) { }
void os_getDevKey (u1_t* buf) { }

static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
// const unsigned TX_INTERVAL = 60;

// Pin mapping
const lmic_pinmap lmic_pins = {
  .nss = 10,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 4,
  .dio = {2, 3, LMIC_UNUSED_PIN},
};

#define resetPin 8

int who;
int AantalSecondeWinterslaap = 20; 

SnoozeAlarm alarm;
SnoozeBlock config_teensy32(alarm);

void onEvent (ev_t ev) {
  Serial.print(os_getTime());
  Serial.print(": ");
  switch (ev) {
    case EV_SCAN_TIMEOUT:
      Serial.println(F("EV_SCAN_TIMEOUT"));
      break;
    case EV_BEACON_FOUND:
      Serial.println(F("EV_BEACON_FOUND"));
      break;
    case EV_BEACON_MISSED:
      Serial.println(F("EV_BEACON_MISSED"));
      break;
    case EV_BEACON_TRACKED:
      Serial.println(F("EV_BEACON_TRACKED"));
      break;
    case EV_JOINING:
      Serial.println(F("EV_JOINING"));
      break;
    case EV_JOINED:
[... 148 lines cut ...]
    
    digitalWrite (21, HIGH);
    Serial.begin(115200);
    Serial.println ("Uit slaap!");
    delay(5000);

    cli();
    hal_sleep_lowpower(AantalSecondeWinterslaap);
    sei();

    os_getTime();   //VERY IMPORTANT after sleep to update os_time and not cause txbeg and os_time out of sync which causes send delays with the RFM95 on and eating power
    //Do here whatever needs to be done after each of the sleepcycle (e.g. check for a condition to break for send or take measurements for mean values etc.)

    os_setTimedCallback(&sendjob, os_getTime() + ms2osticks(10), do_send);  //do a send

    Serial.println(F("go to sleep ... "));
    Serial.println ("delay van 5sec");
    delay (5000);
  }
}

void WinterSlaap (){
  digitalWrite (21, LOW);
  Serial.flush();
  Snooze.hibernate( alarm ); `

matthijskooijman, 2019-07-26:
> @matthijskooijman Thanks, is the hack on millis() working on all platforms?

No, just AVR (others *might* work, but I expect their implementation to be different).

CongducPham, 2019-07-26:
Well, in your program, I believe you should have the call to hal_sleep_lowpower() in the onEvent() function.

Emielcp, 2019-07-26:
@CongducPham same result... :( I tried so many things... damnit

CongducPham, 2019-07-26:
Well, I tested Snooze in standalone with a Teensy32 and it appears that the millis()/micros() counter is preserved. So normally there is no need to change anything to LMIC (not need to insert call to hal_sleep_lowpower()). However, I tested the LMIC example with Snooze on a Teensy and it is not working (the second TX is not scheduled correctly after the sleep). Currently I have no clue on why it is not working. My knowledge of LMIC may be quite limited, maybe Thomas has some idea.

regards,