matthijskooijman/arduino-lmic #228: How to change the Bandwidth

sjaimohd, 2019-05-27:
Hello Sir,

I understand we can change the spreading factor of the code and power transmit but how to change the bandwidth from 125 to 250 or 500? at this line:

  // Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
  LMIC_setDrTxpow(DR_SF12,14);

I checked the radio.c and change the bandwidth from there but there is doesn't show any change in the serial monitor? or it is impossible to change the bandwidth?

haidarm, 2020-01-31:
same here i think the code is limited to 125.
 so its not changeable

adeel1710, 2020-03-12:
Was anyone able to obtain the change of bandwidth ?

raghavyadav11, 2023-05-05:
does anyone has the answer to this?