matthijskooijman/arduino-lmic #230: Issues with changed pinout

sribence, 2019-06-13:
Hi,
we have designed a LoRa board, which contains an ESP32 and an RFM95W.
The problem is, that our pinout is different compared to the board we used before (Heltec).

This is the changed pinout we are using.
`// Pin mapping
#define SS 16 // ESP32 GPIO18 (Pin18) – SX1276 NSS (Pin19) SPI Chip Select Input
#define MOSI 13 // ESP32 GPIO27 (Pin27) – SX1276 MOSI (Pin18) SPI Data Input
#define MISO 12 // ESP32 GPIO19 (Pin19) – SX1276 MISO (Pin17) SPI Data Output
#define SCK 14 // ESP32 GPIO5 (Pin5) – SX1276 SCK (Pin16) SPI Clock Input
#define RST 27 // ESP32 GPIO14 (Pin14) – SX1276 NRESET (Pin7) Reset Trigger Input
#define DIO0 26 // ESP32 GPIO26 (Pin15) – SX1276 DIO0 (Pin8) used by LMIC for IRQ RX_Done & TX_Done
#define DIO1 33 // ESP32 GPIO33 (Pin13) – SX1276 DIO1 (Pin9) used by LMIC for IRQ RX_Timeout
#define DIO2 32 // ESP32 GPIO32 (Pin12) – SX1276 DIO2 (Pin10) not used by LMIC
#define Vext 21

const lmic_pinmap lmic_pins = {
  .nss = SS,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = RST,
  .dio = {DIO0, DIO1, DIO2},
};`

But when we load this code on the device, it stops immediately with a Failure message.
`Starting
Mac: 84:0D:8E:16:8D:E4
FAILURE 
C:\Users\Lab\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\oslmic.c:53`
(This is the full response.)

It's clear, that the problem is new the pinout, but I'm not sure, that:
- the LMIC didn't get the correct pin reference numbers 
- LMIC can't communicate with the RFM95W module, because of some hardware issue (eg. ESP32 does not support such modes on these pinouts)

Thank you for your help.

trlafleur, 2019-06-13:
You may have pin 12 and 13 swapped.....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

Akskaypohankar, 2019-12-10:
hi i am faceing same problem have you found solution yet?

kevin192291, 2020-06-16:
For some reason I am forced to use GPIO15 on the ESP32. If I don't, I get the same error you are getting. It sucks because it eats up another SPI SS pin that I would have much rather used for an SD card.

matthijskooijman, 2020-06-16:
Sounds like there might be some ESP-specific limitations on which pins can be used for what.

Also, it seems you're using https://github.com/mcci-catena/arduino-lmic rather than this version of LMIC, so I suggest you look for support there (that fork is better maintained anyway). So I'm going to close this (but feel free to continue discussion here).

closed 2020-06-16 by matthijskooijman