matthijskooijman/arduino-lmic #233: Dragino lora shield + arduino + 3 sensors works fine but there is some warnings

yassinechamekh, 2019-07-30:
Please help if you have the same problem. 

The warnings are down below 

@matthijskooijman 

C:\Users\ybch\OneDrive - CMR Group\Stage IOT\Projet IOT\Software\Code_Final_Version_2\Code_Final_Version_2.ino: In function 'void setup()':

C:\Users\ybch\OneDrive - CMR Group\Stage IOT\Projet IOT\Software\Code_Final_Version_2\Code_Final_Version_2.ino:263:50: warning: invalid conversion from 'const u1_t* {aka const unsigned char*}' to 'xref2u1_t {aka unsigned char*}' [-fpermissive]

   LMIC_setSession (0x1, DEVADDR, NWKSKEY, APPSKEY);

                                                  ^

In file included from C:\Users\ybch\Documents\Arduino\libraries\arduino-lmic-master\src/lmic.h:5:0,

                 from C:\Users\ybch\OneDrive - CMR Group\Stage IOT\Projet IOT\Software\Code_Final_Version_2\Code_Final_Version_2.ino:2:

C:\Users\ybch\Documents\Arduino\libraries\arduino-lmic-master\src/lmic/lmic.h:305:6: note:   initializing argument 3 of 'void LMIC_setSession(u4_t, devaddr_t, xref2u1_t, xref2u1_t)'

 void LMIC_setSession (u4_t netid, devaddr_t devaddr, xref2u1_t nwkKey, xref2u1_t artKey);

      ^

C:\Users\ybch\OneDrive - CMR Group\Stage IOT\Projet IOT\Software\Code_Final_Version_2\Code_Final_Version_2.ino:263:50: warning: invalid conversion from 'const u1_t* {aka const unsigned char*}' to 'xref2u1_t {aka unsigned char*}' [-fpermissive]

   LMIC_setSession (0x1, DEVADDR, NWKSKEY, APPSKEY);

                                                  ^

In file included from C:\Users\ybch\Documents\Arduino\libraries\arduino-lmic-master\src/lmic.h:5:0,

                 from C:\Users\ybch\OneDrive - CMR Group\Stage IOT\Projet IOT\Software\Code_Final_Version_2\Code_Final_Version_2.ino:2:

C:\Users\ybch\Documents\Arduino\libraries\arduino-lmic-master\src/lmic/lmic.h:305:6: note:   initializing argument 4 of 'void LMIC_setSession(u4_t, devaddr_t, xref2u1_t, xref2u1_t)'

 void LMIC_setSession (u4_t netid, devaddr_t devaddr, xref2u1_t nwkKey, xref2u1_t artKey);

      ^

Le croquis utilise 23202 octets (71%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
Les variables globales utilisent 929 octets (45%) de mémoire dynamique, ce qui laisse 1119 octets pour les variables locales. Le maximum est de 2048 octets.

matthijskooijman, 2019-07-30:
These warnings are harmless, but should probably be fixed at some point. They might be fixed already in LMIC 1.6, or the https://github.com/lorabasics/basicmac stack based on LMIC, so just ignore them here for now. I'll leave this issue open in case anyone else wonders.

yassinechamekh, 2019-07-31:
@matthijskooijman please where can i find the latest updated LMIC library ? Thank you.

matthijskooijman, 2019-08-02:
I can't help you with that, I haven't ported the updated 1.6 library yet (though I have had plans for years). I think there are some forks of this repo around that do have the newer version. I did look into basicmac at some point, but haven't managed to make that work on Arduino yet (also, it seems to have grown too big for AVR).

trlafleur, 2019-08-02:
It would be more effective to build an Arduino port of the current Semtech
base code that will soon be LoRaWan 1.1 compliant... it's an up to date
derivative of the IBM LMiC 1.6 codebase...
will also feature firmware updates over LoRaWan

https://github.com/Lora-net/LoRaMac-node

matthijskooijman, 2019-08-02:
@trlafleur, AFAIU LoRaMac-node is *not* related to LMIC in any way, it's a completely separate codebase.

BasicMac (as I linked above) OTOH, *is* based on LMIC and should (almost?) 1.1 compliant, but my first attempt at porting show that it's also a lot bigger than LMIC 1.5/1.6 (i.e. probably won't fit into a 328p).

trlafleur, 2019-08-02:
its time for people to move on, 32bit processors like the M0, are cheaper,
faster, less power that 8bit 328P... my two bits....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

yassinechamekh, 2019-08-23:
@matthijskooijman  problem solved.
You only have to cast it and the compilation will past with out any warning.
LMIC_setSession (0x1, DEVADDR, (xref2u1_t)NWKSKEY, (xref2u1_t)APPSKEY);

matthijskooijman, 2019-08-23:
Ok, thanks for reporting back. I'm closing this issue as solved :-)

closed 2019-08-23 by matthijskooijman

prajwalavian, 2020-07-06:
@yassinechamekh  i have the same warning please help me