matthijskooijman/arduino-lmic #236: Using SPI with I2S ends up with multiple def

fontanon, 2019-08-05:
Hi everybody !!

I've an Adafruit Feather M0 board and I'm trying, on the one hand, to capture info from a I2S mic sensor (specifically Invensense INMP441[1]), and on the other, to report via LoRaWAN (and The Things Network). 

[1] https://www.invensense.com/wp-content/uploads/2015/02/INMP441.pdf

The firmware is pretty basic. Using Arduino IDE + Adafruit's samd 1.5.2 board libraries[2] (to interact with the board and sensor) + arduino-lmic[3] (to interact with RFM95). Error comes when compiling. **My interpretation is both arduino-lmic and Adafruit's samd 1.5.2 board libraries end up defining DMAC_Handler, so a multiple definition error raises.**

[2] https://learn.adafruit.com/adafruit-feather-m0-basic-proto/using-with-arduino-ide
[3] https://github.com/matthijskooijman/arduino-lmic

```shell
/tmp/arduino_build_271823/libraries/Adafruit_ZeroDMA/Adafruit_ZeroDMA.cpp.o: In function `DMAC_Handler':
Adafruit_ZeroDMA.cpp:(.text.DMAC_Handler+0x0): multiple definition of `DMAC_Handler'
/tmp/arduino_build_271823/libraries/I2S/utility/DMA.cpp.o:DMA.cpp:(.text.DMAC_Handler+0x0): first defined here
collect2: error: ld returned 1 exit status
```

I think this is because **arduino-lmic hal library includes SPI which defines DMAC_Handler as well as I2S's DMA library.**

I'd appreciate any help. Are arduino-limic and Adafruit's samd 1.5.2 board libraries compatible? Is there anything I could do to bypass the DMAC_Handler multiple definition error?

This is pretty easy to reproduce. You do not even need a physical Adafruit Feather M0, just download the libraries and try compiling. Find a referece arduino sketch below:

```c++
#include <Arduino.h>
#include <I2S.h>
#include <lmic.h>
#include <hal/hal.h>
#include <stdio.h>
#include <math.h>

void setup() {
}

void loop() {
}
```

matthijskooijman, 2019-08-05:
If you analysis is correct, then the problem is that the SPI library and the I2S library are incompatible with each other, and this problem is not really related to LMIC (to test, you could change `lmic.h` to `SPI.h` and remove the `hal/hal.h` include in your testcase and see if it still breaks.

Looking at the actual error, however, it seems that the conflict is between the Adafruit_ZeroDMA library and the I2S library. I know that there is not currently any standardized DMA library for the Arduino/Adafruit zero boards, which could lead to conflicts (see also https://github.com/arduino/ArduinoCore-samd/issues/419 I reported, which seems to point out exactly the incompatibility you see as well).

I'm not entirely sure where to go with your issue (or whether a workaround is available, I'm not familiar with the actual code itself), but I'm pretty sure the issue is not here (so I'm closing this one). Maybe you could report at the Adafruit_ZeroDMA repository, though a proper fix probably needs something from Arduino...

closed 2019-08-05 by matthijskooijman

fontanon, 2019-08-05:
Hi @matthijskooijman, thanks for your answer. You're right, just replacing lmic/hal with SPI the same error is reproduced. I agree with you this is not an arduino-lmic related issue. 

I'll follow your report on arduinocode-samd and I'll open an issue on adafruit_zerodma as well to let them know. I guess there should be a possible implementation so _DMAC_Handler_ is not defined twice when using I2S and SPI for Adafruit M0 with arduinocore-samd.

```c
#include <Arduino.h>
#include <I2S.h>
#include <SPI.h>
#include <stdio.h>
#include <math.h>

void setup() {
}

void loop() {
  
}
```
```shell
/tmp/arduino_build_458350/libraries/Adafruit_ZeroDMA/Adafruit_ZeroDMA.cpp.o: In function `DMAC_Handler':
Adafruit_ZeroDMA.cpp:(.text.DMAC_Handler+0x0): multiple definition of `DMAC_Handler'
/tmp/arduino_build_458350/libraries/I2S/utility/DMA.cpp.o:DMA.cpp:(.text.DMAC_Handler+0x0): first defined here
collect2: error: ld returned 1 exit status
```