matthijskooijman/arduino-lmic #237: Not working with Arduino MKRZero

tushar-semwal, 2019-08-13:
I tried uploading on MKR Zero but failed to do so. The program compiles however fails to upload.

matthijskooijman, 2019-08-13:
Bummer. I rather doubt this is caused by this particular library, but maybe. However, if you're expecting any help, you will have to provide more details (e.g. error messages, tell us if you tried other sketches, etc.).

tushar-semwal, 2019-08-13:
The steps I followed:
1. Downloaded the zip file of the library and installed it on Arduino.
2. It compiles successfully. See image below:
[image]
3. However, If I upload, it verifies and suddenly breaks the MKR. The board gets disconnected and does not get detected any more. I need to do a double reset to make it get recognised again.
[image]
4. I am using this arduino: The new MKR zero: https://store.arduino.cc/arduino-mkrzero

matthijskooijman, 2019-08-13:
Ah, I suspect this is because some assertion fails, which prints a message, disables interrupts and locks up the MCU. Normally, on AVR this is fine, since a new upload will reset the MCU. However, on native USB boards such as the Leonardo or the MKR Zero, the board is reset by setting a specific baudrate on the serial port, which needs interrupts enabled.

You could try removing this line: https://github.com/matthijskooijman/arduino-lmic/blob/ba1265d5d2f775177cdc7c82186724e4f0bdc3a8/src/hal/hal.cpp#L251

I suspect that should make uploads work again.

Then there is still the failure that actually causes this, which is printed to serial. However, since the board is not reset when you open the serial port, it is likely that you'll miss the message (by the time you open the serial port, the message has already been printed). You could try including a `while (!Serial);` line in the `setup()` function (after `Serial.begin(...)`), which will make the board wait for the serial port to be opened before continuing (remember to remove the line again when you want to run the board without the serial port open).

Also note that I've never tested this library on SAMD boards like the Zero, so there might be other problems as well.

tushar-semwal, 2019-08-13:
Thanks @matthijskooijman for a quick solution. I was able to upload the code. However, once I open the serial terminal, and as you suspected, I got the failure printed. Also, I tired putting a print statement on the loop(), but it is  not printing. The output is:
```Starting
FAILURE 
C:\Users\tsemwal\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
```
Though I have not yet connected the HopeRFM95W lora module. I am just testing it on the bare arduino but should it matter?

matthijskooijman, 2019-08-13:
> Though I have not yet connected the HopeRFM95W lora module. I am just testing it on the bare arduino but should it matter?

Yes, on startup the library reads the LoRa module's model number and if it does not match the expected value, it prints this FAILURE and stops executing, so that's your problem.

I'm going to go ahead and close this issue, since it seems to be clear how things are now, but feel free to add more comments or reopen if needed.

closed 2019-08-13 by matthijskooijman

tushar-semwal, 2019-08-14:
Thank you so much @matthijskooijman . Really grateful for the quick solution.